OctoPrint/OctoPrint #1379: Cancelling a print should NOT cause the Z axis to home

cweagans, 2016-06-18:
#### What were you doing?

I was printing something and the print failed, so I hit the cancel button. Extrusion stopped and the head started to move toward 0,0,0.

#### What did you expect to happen?

Z should increase a bit immediately, then zero X

#### What happened instead?

The head started moving toward 0,0,0 causing the extruder assembly to crash into the partially printed part.

#### Branch & Commit or Version of OctoPrint

Version: 1.2.13 (master branch)

#### Printer model & used firmware incl. version

Monoprice Maker Select/Wanhao Duplicator i3 -- whatever firmware it shipped with.

#### Browser and Version of Browser, Operating System running Browser

n/a, but Chrome on Mac

I have read the FAQ.

gddeen, 2016-06-18:
Could the Buttons have been clicked twice? 
Could someone else have been printing?
Did you already have a print on the tray, start a new print, then click cancel?

I Cancel a LOT. At least 3 times for every print... I use Chrome from Mac always...

gddeen, 2016-06-18:
Did you print something, unplug the printer (big ball of mess), plug it back in, then cancel?

cweagans, 2016-06-18:
I'm the only one that uses this printer. There is no possibility of anyone else using it. I started printing an object a 9am. At 11am, I noticed layer delamination. At 11:01, I clicked the cancel button and the fan assembly crashed into my print on the way to 0,0,0.

Could my gcode script have something to do with it? I don't see anywhere that I'm homing Z or Y, but maybe I'm missing something?

Here are the scripts I have set up:

**After print job completes**:

```
; Disable all heaters
{% snippet 'disable_hotends' %}
{% snippet 'disable_bed' %}

; Retract the filament 1mm
G1 E-1 F300

; Move Z up, extruder away, and retract 3mm
G1 Z+0.5 E-3 X-20 Y-20 F6000

; Fan at 100% to cool nozzle
M106 S255

; Absolute positioning
G90

; Home X to get it out of the way.
G28 X0

; Steppers off
M84

; Wait 60 seconds
G4 P60000

; Fans off
M106 S0
```

**After print job is cancelled**:

```
; Retract the filament 1mm
G1 E-1 F300

; Move Z up, extruder away, and retract 3mm
G1 Z+0.5 E-3 X-20 Y-20 F6000

; Fan at 100% to cool nozzle
M106 S255

; Absolute positioning
G90

; Home X to get it out of the way.
G28 X0

; Move bed forward
G28 Y190

; Steppers off
M84
```

gddeen, 2016-06-18:
I've done that one and it was bizarre. I had expected the "Nx" to stop it... I guess it had picked up in the middle of a command line.

gddeen, 2016-06-18:
Oh yes... There is a Z0.5 ... I'd add a G91 to make sure you are in relative mode.

gddeen, 2016-06-18:
At the start of both of those scripts...

foosel, 2016-06-18:
My money is on what @gddeen said. Slicers usually generate gcode in absolute mode. And since there's nothing in your cancel script that switches to relative mode that z+0.5 doesn't move the z axis up a bit but moved it to absolute coordinate 0.5

Add a G91 at the very beginning of both those scripts and the issue should vanish ;)

gddeen, 2016-06-18:
Sweet! I'm not a complete idiot like my wife says! OK, well maybe I am...

ntoff, 2016-06-20:
Also why are you G28'ing to get Y to move to 190? Shouldn't that be a G1 Y190 ?

gddeen, 2016-06-20:
**After print job completes:**

```
; Switch to relative coordinates
G91

; Disable all heaters
{% snippet 'disable_hotends' %}
{% snippet 'disable_bed' %}

; Retract the filament 1mm
G1 E-1 F300

; Move Z up, extruder away, and retract 3mm
G1 Z+0.5 E-3 X-20 Y-20 F6000

; Fan at 100% to cool nozzle
M106 S255

; Absolute positioning
G90

; Home X to get it out of the way.
G1 X0

; Steppers off
M84

M117 "Thanks Gina"
; Wait 60 seconds
G4 P60000
M117 "          "

; Fans off
M106 S0

````
**After print job is cancelled:**

```
; Switch to relative coordinates
G91

; Retract the filament 1mm
G1 E-1 F300

; Move Z up, extruder away, and retract 3mm
G1 Z+0.5 E-3 X-20 Y-20 F6000

; Fan at 100% to cool nozzle
M106 S255

; Absolute positioning
G90

; Home X to get it out of the way.
G1X0

; Move bed forward
G1Y190

; Steppers off
M84
M117Cancelled
```

foosel, 2016-07-21:
@cweagans have you tried the suggested `G91` solution?

cweagans, 2016-07-21:
Ah, yeah. Sorry. Thought I updated this issue. All good. Sorry for the noise!

closed 2016-07-21 by cweagans