OctoPrint/OctoPrint #1385: M114 output (related to issues 1373)

MoonshineSG, 2016-06-22:
#### What were you doing?

Below mentioned commit causes issues with plugins which process lines output line from M114 to read the current XYZ values... 

The output from M114 doesn't get add to "messages" in "current" 

#### What did you expect to happen?

received M114 output

#### What happened instead?

received empty data because comm.py#L1050

```
if 'X:' in line and 'Y:' in line and 'Z:' in line:
    pass
```

#### Branch & Commit or Version of OctoPrint

https://github.com/foosel/OctoPrint/commit/ad201b040d3d76a8de8177de563deaff20894476

#### Printer model & used firmware incl. version

marlin (custom)

#### Browser and Version of Browser, Operating System running Browser

n/a

#### Link to octoprint.log

not needed

#### Link to contents of terminal tab or serial.log

not needed

#### Link to contents of Javascript console in the browser

not needed

#### Screenshot(s) showing the problem:

not needed

I have read the FAQ.

foosel, 2016-07-14:
Fixed in `maintenance` and `devel` branches.

foosel, 2016-07-28:
Merged to `master` and released with 1.2.14

closed 2016-07-28 by foosel