OctoPrint/OctoPrint #1387: MQ-2 sensor / Fire safety plugin

Mystereon, 2016-06-23:
Please could we implement an MQ-2 gas sensor plugin ??? I'm rather worried about potential fire issues if I leave the printer unattended overnight. I've been readings some rather frightening reports of stray heater elements due to swapouts and lazy grub screw torsion policy. My head has been spinning with designs for inert gas release systems and alarms, both quite relavent to extinguishing a printer blaze. The sensors are ridiculously cheap and the code exists for smoke detectors , hardly a processor drain when enabled in my honest opinion.. I'm keen to hear hear peoples views on this. Although I have not experienced a printer fire, thankfully , I have experienced the element coming loose when I could swear I had tightened it fully.

masto, 2016-09-25:
I saw a photo of someone's cabinet with a couple of these firestop cans mounted in the top: http://amzn.to/2cDfjwk. I just ordered some to put in my enclosure.

Smoke sensor is a nice idea too, for remote monitoring and alerting.

markwal, 2016-09-25:
If you wanted, you could wire up a smoke detector to the gpio pins and write a python program to monitor them and do "something" when it triggers.

I've got a roll my own set-up with a relay that allows me to interrupt all power to the printer from the mains.  I just have it hooked up to an overtemp alert from the printer firmware to catch the case when one of the mosfets fails closed.  However, cutting the power when there is already a fire won't do much beyond make it easier to put out by an extinguisher.  So I also have an extinguisher nearby.  I also bought an automatic one to hang over the top, but haven't installed it yet.

jneilliii, 2022-01-23:
Curious if the enclosure plugin could be an option for this interaction?

jneilliii, 2022-01-23:
Based on reading the Enclosure description and a couple of closed issues, this is possible with Enclosure plugin, which was released after last comment, closing.

closed 2022-01-23 by jneilliii