OctoPrint/OctoPrint #1388: Auto reconnect after turning the printer off and on.

MichMich, 2016-06-23:
#### What were you doing?

When a print on my Lulzbot Mini is finished I take it off the built plate and turn off the printer. The Raspberry Pi (Running OctoPrint) remains on. If I want to print something new, I turn on the printer, and open OctoPrint in my browser. OctoPrint then still thinks it's connected, but the communication with the printer doesn't work until I click "Disconnect" and reconnect after disconnecting using the "Connect" button.

#### What did you expect to happen?

I expect OctoPrint to automatically try to reconnect with the printer if it stops responding. As soon as I turn on the printer, the connection should be reestablished. 

#### What happened instead?

OctoPrint says it's connected, but all commands fail. The printer does not respond until reconnecting.

#### Branch & Commit or Version of OctoPrint

Version: 1.2.13 (master branch)

#### Printer model & used firmware incl. version

Lulzbot Mini. (Firmware unknown, will look into this after my print finishes.)

#### Browser and Version of Browser, Operating System running Browser

Safari 9.1.1 (But the browser has no influence on this.)

#### Link to octoprint.log

http://pastebin.com/3gbJ1QrH

#### Link to contents of terminal tab or serial.log

Printer is currently printing. Will add serial log when the problem occurs. (Next time I print)

#### Link to contents of Javascript console in the browser

NA

#### Screenshot(s) showing the problem:

NA

foosel, 2016-06-23:
Bot complained because you didn't copy the template completely...

Telling it to step down but marking the ticket as "awaiting information" since this a problem that definitely needs a `serial.log` or at the very least the output from the terminal tab to have any change to diagnose further.

nophead, 2016-06-23:
I have found that more recent OctoPrint versions need a fake ack after
switching the printer on and off.

KC703, 2016-06-24:
I also use the Fake Ack under the Terminal->Advanced to re-connect to the printer after shutting it off / on remotely.

MichMich, 2016-07-01:
I've added the Serial log to the initial issue. 

I can confirm that the Fake Ack is a solution, but it would be better if this wasn't nessecery.

foosel, 2016-07-02:
```
[Printer Turned Off/On]

Changing monitoring state from 'Operational' to 'Printing'
```

That looks like *immediately* after turning the printer back on you directly started printing - is there something missing from the log or is that really what happened here?

Or am I simply misreading that log annotations by you?

It also looks a bit like while powering off the printer lead to the firmware not replying anymore, the serial connection was nevertheless still kept open. Is that indeed the case?

MichMich, 2016-07-02:
No, after turning the printer back on, there simply isn't any communication. Even if I wait 10 minutes before I press print, it wont work. I need to do a Fake Ack and then the communication will work again.

foosel, 2016-07-02:
But while the printer is turned off there is also no communication, even though the serial connection is still present?

MichMich, 2016-07-02:
Exactly.

bolsoncerrado, 2016-07-19:
Not sure if related, mine is not a "branded" printer, but i've noticed since recent updates that Autoconnect on startup no longer works?

foosel, 2016-07-19:
@bolsoncerrado huh, that works just fine here, both with a real printer and the virtual printer, on master (1.2.13) and maintenance branch (1.2.14.dev).

The issue described by @MichMich rather seems to be caused by the fact that OctoPrint never picks up on the printer being AWOL since the serial connection never drops on its own but instead just stops responding with `ok`. OctoPrint currently ignores serial timeouts while not printing, so I guess that has to be changed in order to fix this. That will still not allow OctoPrint to immediately detect that the printer went away if it just stops responding without dropping the serial line - I can only work with timeouts there sadly, the printer not responding because it decided to die reads exactly the same as a printer which is currently busy with an extra long move or a printer which send an ok just right but it got garbled on the serial line.

Been trying to fix this today but ran into some odd deadlock issues I don't fully understand so far.

MichMich, 2016-07-19:
Hi Gina, the fake acknowledgement is a workable workaround for the time being. So take your time looking for a resolution.

foosel, 2016-07-28:
It's "solved" now in the way that at least OctoPrint will now again detect printers that just go AWOL with any life sign, but still keep the serial line (which they don't reply on anymore) open. I've added a tracking of consecutive communication timeouts, once a certain limit is reached (individually configurable - via `config.yaml` only for now - for idle, printing and during long running commands) the printer will be considered gone and the connection closed.

That means you will have to reconnect though, which is better than having to press semi-hidden buttons or disconnect AND reconnect however, and it better mirrors the actual state of the connection - the printer is no longer really connected when it doesn't respond on serial anymore.

foosel, 2016-07-28:
> via `config.yaml` only for now

Correction, I in fact did add those settings to the serial configuration dialog and just forgot about it O_O

[image]

MichMich, 2016-07-28:
Great, I'll give it a try!

foosel, 2016-07-28:
Merged to `master` and released with 1.2.14.

closed 2016-07-28 by foosel

MichMich, 2016-08-06:
@foosel I can confirm this issue is solved in the latest release! Thanks!