OctoPrint/OctoPrint #1390: Error in config.yaml problem

dagzjenkins, 2016-06-25:
Seems like I have a bad value in a config.yaml file. When running 

~/OctoPrint/venv/bin/octoprint

I keep receiving the error:

yaml.parser.ParserError: while parsing a block mapping in "/home/pi/.octoprint/config.yaml", line 1, column 1 expected <block end>, but found '<block sequence start>' in "/home/pi/.octoprint/config.yaml", line 5, column 2
Full config file can be found here:
http://pastebin.com/vf2PTgd3

Any help would be greatly appreciated!!!

Additional detail: Pi 3 running w/ Wanhoa Duplicator i3

nophead, 2016-06-25:
You have a stray > on line 2:
```
stream: http://192.168.1.144>:8080/?action=stream
```

It should be:
```
stream: http://192.168.1.144:8080/?action=stream
```

I don't know why you got the error message you did though unless something else is wrong.

ntoff, 2016-06-26:
Watch out if you're copy/pasting from windows. I've had issues with CR+LF creeping into stuff when editing with nano via SSH, those will royally screw everything up. I think I only found this out after opening something up in vi and noticing a bunch of extra stuff.

foosel, 2016-06-29:
You should have gotten your answer with what @nophead wrote.

Questions like this are actually better suited for the mailinglist or the G+ community than the bug tracker - it's not a bug :)

Closing this.

closed 2016-06-29 by foosel