OctoPrint/OctoPrint #1392: serial connection not working anymore under 1.2.13

use-r42, 2016-06-28:
After updating to version 1.2.13 (master branch) OctoPrint cannot connect to the printer (Velleman K8200) anymore.
After trying to connect with the baud rate set to "auto" I am getting the following error:
"Unhandled communication error
The was an unhandled error while talking to the printer. Due to that OctoPrint disconnected. Error: No more baudrates to test, and no suitable baudrate found."

Downgrading the Octoprint version using the image "2016-03-18-octopi-jessie-lite-0.13.0.img" solves the problem. But upgrading to 1.2.13 creates the problem again.

markwal, 2016-06-28:
This looks like a support request for the Google Group or the Google Plus Community linked in the contribution guidelines.  When you post there, please also include what happens when you select a baud rate (115200, for example).  Thanks.

foosel, 2016-06-30:
I'm not entirely sure if that is strictly a support request issue since updating changes the observed behaviour, but this ticket doesn't contain enough information to go on with any attempt to resolve the problem at hand.

I'm reopening this for now, but @use-r42, you'll need to provide some more data to go on. Most importantly log files - especially a `serial.log` from a connection attempt, the contribution guidelines linked above tell you how to obtain this - and precise information on the version(s) of OctoPrint with which auto detection of the serial speed still worked for you.

According to the commit history I didn't touch anything with regards to the auto detect code between 1.2.10 (what OctoPi 0.13 ships with) and 1.2.13, but it might be a side effect of some other change. To narrow that down though more data is needed.

ntoff, 2016-07-01:
Raspberry pi 3
Jessie Linux 4.4.9-v7+ #884 (ignore whatever this auto hashtag link thing creates, 884 is the build number? I think? of the kernel? I got it from uname -a)
Octoprint Version: 1.2.13 (master branch)
3drag/k8200 board
Marlin firmware v2 (their own version numbering, v1 was sans LCD, v2 is for LCD controller support)

Installed manually via github and it connects just fine so if there is an issue it might only be with the octopi image versions.

foosel, 2016-07-01:
@ntoff that's a nice reassurance then that it's not reproducible with a manual github install. 

Still pretty much unable to make a competent analysis of the OP's issue though with their collaboration with regards to the required data... :/

ntoff, 2016-07-01:
I did manually set the baud rate though so maybe the auto thing is what did it. I didn't really go into it too much because it's an install I'd been messing about with so I've just thrown the octopi image on an SD and will test a bit more with both versions of velleman's firmware and leave the baud rate on auto. Anything else you'd like?

foosel, 2016-07-01:
If it fails with auto, the `serial.log` from that :D

ntoff, 2016-07-01:
Well, I just set it to auto for usb and baud and sure enough it *found* the serial port but wouldn't connect to it. Auto on the baud rate but manual on the usb device connected fine. That's on the stock octopi image of 1.2.10 though. Serial log isn't exactly huge so here's all that's in it.

>2016-07-01 14:58:52,527 - SERIAL - DEBUG - Enabling serial logging
2016-07-01 14:59:05,336 - SERIAL - DEBUG - Enabling serial logging
2016-07-01 15:00:04,906 - SERIAL - DEBUG - Changing monitoring state from 'Offline' to 'Detecting serial port'
2016-07-01 15:00:04,929 - SERIAL - DEBUG - Serial port list: ['/dev/ttyUSB0']
2016-07-01 15:00:04,943 - SERIAL - DEBUG - Connecting to: /dev/ttyUSB0
2016-07-01 15:00:06,258 - SERIAL - DEBUG - Error while connecting to /dev/ttyUSB0: 'Timeout'
2016-07-01 15:00:06,262 - SERIAL - DEBUG - Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
2016-07-01 15:00:06,263 - SERIAL - DEBUG - Failed to autodetect serial port, please set it manually.

foosel, 2016-07-01:
Serial port auto detection only works with Arduino compatible devices, so I'm not too surprised if that's failing. But auto detection of the baudrate doesn't rely on some programmer mode being possible, so that should usually work. Can you try that too?

ntoff, 2016-07-01:
Baud rate was on auto, I only changed the serial port between auto and manual. It automatically found the baud rate ok and the 3drag board is an ATMega2560 based board with an FTDI usb-serial interface so doesn't that mean it's an arduino compatible board? Also here's the serial log for a *manual device - auto baud rate* test https://gist.github.com/ntoff/5dc4a758dbef9a1f24dd1c057071f4c1

foosel, 2016-07-01:
> the 3drag board is an ATMega2560 based board with an FTDI usb-serial interface so doesn't that mean it's an arduino compatible board?

I phrased that poorly. Thing is, the port auto detect tries to detect a printer by putting it in programming mode. For a lot of arduino based machines that works fine. Some do not allow that without some pin shorted though. To be honest, that auto detection code is one of the last things still inherited from Cura, and I've been meaning to replace it for a long time now with something a bit more solid, but I simply have not found the time yet. Just trying to connect all available serial ports with all available baudrates would probably take a long time on first connect, but still better than it failing just because the board doesn't allow programming mode. 

Regarding the actual issue at hand here though (baud rate detection having stopped working after update), your linked log file above looks just like it should look - iteration over common baud rates, testing if something looking like a printer response comes back, if so, yay, success.

Anyhow... marking this as unreproduced for now on top of "awaiting information" and giving @use-r42 some more time to get back with the missing data from their attempts.

ntoff, 2016-07-01:
Sorry, had to go help someone fix some email stuff. Here's the log from the auto updated 1.2.13 version: https://gist.github.com/ntoff/87608378793ad28ade2b59171d3726ee

Pretty much the same thing. Auto baudrate and manual device connects fine.

Also you're right, the 3drag board has a jumper that disables the reset line so connecting to it doesn't cause an automatic reset.

foosel, 2016-07-16:
No reply from OP after two weeks. Since it worked as expected on a test against the same printer board @ntoff, assuming this working. Please request reopen *with a serial.log showing the problem* of it isn't after all.

closed 2016-07-16 by foosel