OctoPrint/OctoPrint #1393: API should allow pause and unpause, not just pause-toggle

MarcusWolschon, 2016-06-30:
The API
http://docs.octoprint.org/en/master/api/job.html#issue-a-job-command
allows for a "command": "paus" described as:
"Pauses/unpauses the current print job."

Issue:
Issuing a pause command twice for any reason (timeout, response not parsed properly, script was restarted,..) resumes the print instead of keeping it paused.

In my case I'm using the pause command for a Filament monitor.
http://marcuswolschon.blogspot.de/2016/06/ultimaker-ii-filament-sensor-and-remote.html
Obviously having any chance of accidentally resuming a print instead of pausing it, is dangerous.

Proposal:
The current command is nice for user interfaces used exclusively by humans.
For automatic scripts it should offer 2 command to
pause and to resume a print regardless of state.

MarcusWolschon, 2016-06-30:
Relevant code:
https://github.com/foosel/OctoPrint/blob/master/src/octoprint/server/api/job.py#L40
https://github.com/foosel/OctoPrint/blob/685ce16c85eea9c4f11586ea99c9a84aca382bd4/src/octoprint/printer/standard.py#L407
https://github.com/foosel/OctoPrint/blob/692166f067329cd3d6fdc84389e0dd76184c5e0c/src/octoprint/printer/__init__.py#L239

Documentation to update:
https://github.com/foosel/OctoPrint/blob/692166f067329cd3d6fdc84389e0dd76184c5e0c/docs/features/action_commands.rst

Js code only needs to be updated if the user interface would make use of it, does it?

foosel, 2016-06-30:
Implemented in `maintenance` and `devel` branch, will be part of next release.

Simply changing the behaviour of the `pause` command and introducing a `resume` and `toggle` command was not possible due to reasons of backwards compatibility (there might be third party clients depending on the toggle behaviour). Instead a new optional `action` parameter was added to the existing `pause` command which can be either `pause`, `resume` or `toggle`. The latter will be assumed if the `action` parameter is not present. API docs are adjusted accordingly.

closed 2016-06-30 by foosel