OctoPrint/OctoPrint #1394: Don't count warmup time as print time.

tenten8401, 2016-06-30:
#### What were you doing?

Printing

#### What did you expect to happen?

Print time to be calculated excluding warm up time

#### What happened instead?

Print time was calculated with warm up time

#### Branch & Commit or Version of OctoPrint

Version: 1.2.13 (master branch)

#### Printer model & used firmware incl. version

PrintrBot Simple Metal (Firmware v5?)

I have read the FAQ.

gddeen, 2016-07-01:
I count each print time as how long it takes. So... I really don't like that the warmup time would be excluded.

That's like selling something for $299.99 and finding out it is really $364.32 with taxes..

tenten8401, 2016-07-01:
Then at least add the warmup time to the estimated time remaining. Causes about 2 minutes off on the ETA and it's annoying as crap. :(

Salandora, 2016-07-01:
Well the ETA isnt even precice.
The actual ETA calculation e.g. doesn't use jerk or acceleration. (Because of so many different types of implementation even with this it would be a rough guess)
So your 2 minutes off could also be because of this.
I'm actually working on another calculation approach as an seperate programm, which shall implement a way to calculate for Marlin, Repetier and a Generic variant.
It will also support a method to take warmup time into account.
Either by a user defineable value or by heating things up and taking the needed time.

foosel, 2016-07-01:
The heatup time [should actually already be substracted for the running estimate](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/printer/standard.py#L616). When OctoPrint sends an `M109` or `M190` it will [set an internal `heating` flag](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/util/comm.py#L2037), likewise when it [sees a temperature line without an `ok` unless that's disabled](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/util/comm.py#L1060), and [accumulate the time spent waiting for the firmware to heat up](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/util/comm.py#L1259).

That will influence what "Print Time Left" will report. Which is still inaccurate due to the reasons stated above.

Or are you talking about the "Approx. Total Print Time" value? That one is strictly based on the GCODE and I can't make any assumptions about the heatup time in there since I don't know how long it will take, too many environmental factors at play (your hotend could already be at temperature when you click print, or it could be freezing in your room, or there could be draft).

tenten8401, 2016-07-01:
So what I've noticed, is that the "Print Time Left" is always about 2 minutes off as the nozzle takes 2 minutes to heat, so it should not be counting down the print time left unless the nozzle and bed is done heating. What I also noticed is that it does not count when the bed is heating.

-Isaac

closed 2016-07-01 by tenten8401