OctoPrint/OctoPrint #1401: [BUG] You should consider upgrading via the 'pip install --upgrade pip' command -> WHAT PIP!?

bolsoncerrado, 2016-07-08:
#### What were you doing?

Upgrading plugins

#### What did you expect to happen?

Plugin upgrade

#### What happened instead?

Worked fine

#### Branch & Commit or Version of OctoPrint

master 1.2.13

#### Browser and Version of Browser, Operating System running Browser

FireFox 47.0 

#### Link to octoprint.log

2016-07-01 18:24:20,426 > Installing collected packages: OctoPrint-SnapStream
2016-07-01 18:24:20,427 > Running setup.py install for OctoPrint-SnapStream: started
2016-07-01 18:24:23,433 > Running setup.py install for OctoPrint-SnapStream: finished with status 'done'
2016-07-01 18:24:23,434 > Successfully installed OctoPrint-SnapStream-0.2.3
2016-07-01 18:24:25,404 ! You are using pip version 8.1.1, however version 8.1.2 is available.
2016-07-01 18:24:25,405 ! You should consider upgrading via the 'pip install --upgrade pip' command.

#### Issue:

Upon going to SSH my pi:

pi@case:~ $ pip install --upgrade pip
-bash: pip: command not found

I have read the FAQ.

So not sure which part is the problem!?

foosel, 2016-07-08:
That message is generated by Python's own pip command. There's nothing I can do to make it include a path to the specific version of pip you need to use, or to suppress it. `pip` is not written by me, it's a core Python tool used to install Python packages, which OctoPrint's plugins are as well. Not everything that gets output in red is an error or even a problem, so no need to be alarmed here and throw punctuation at me :)

On OctoPrint it would be `~/oprint/bin/pip install --upgrade pip` what you need to run to get rid of that message. If you installed manually you'll need to use the `pip` command belonging to the same python instance you installed OctoPrint under (you should know best what that one was). The message can also just be ignored however.

> ### What happened instead?
> Worked fine

I had to chuckle.

> [UnicodeWarning]

No idea about that (missing context), but it's unrelated but apparently not causing any issues either since you stated yourself that everything works fine.

bolsoncerrado, 2016-07-08:
Thanks on that. Im on recent Jessie using Octopi's IMG.

closed 2016-07-08 by bolsoncerrado