OctoPrint/OctoPrint #1402: [Request] Add hotend and bed temperatures to state panel

mabrown, 2016-07-08:
After printing, I always allow the hotend to cool before turning off the fan to help prevent jams due to overheating filament in the cold end. Having the hotend temperature on the "State" panel, which is visible no matter which UI tab you're on, would be a nice add because it would reduce the number of interactions I need to perform to get the job done.
Workflow now:

1. Begin print
2. Switch to GCode Viewer tab to watch the fun
3. Wait for the print to finish
4. Switch to the Temperature tab until the hotend temp drops to the desired level
5. Switch to the Control tab
6. Click the Fan off button 

Desired workflow:

1. Begin print
2. Switch to GCode Viewer tab
3. Wait for the print to finish
4. Wait for the hotend temp to drop,monitoring via the new "temp" line on the "State" panel
5. Switch to the Control tab
6. Click the Fan off button

I'm using version 1.2.13 (master branch) with a Monoprice Select Mini 3D printer (firmware version unknown). I monitor from a Windows 10 PC using the MS Edge browser. Sometimes from my Windows 10 phone using the same.

markwal, 2016-07-08:
What UI are you using from the Windows phone?  (ie are you using TouchUI plugin)

For the desktop this would do your scenario though it uses the bar at the top rather than the state panel.  It would be easy to make a plugin that added a panel that is always there for temperatures though: http://plugins.octoprint.org/plugins/navbartemp/

mabrown, 2016-07-08:
I do use the TouchUI plugin on the phone. The Navbar Temp plugin is perfect! I don't know how I missed it.

Salandora, 2016-07-08:
You could also automate this process like this:

```
M109 S30
M106 S0
M109 S0
```

Into your end code

mabrown, 2016-07-08:
Salandora, let me see if I understand what that command sequence does. M109 S30 tells the machine to set the temp to 30C and holds until it reaches that temp, then M106 S0 turns off the fan (sets its speed to 0), then M109 S0 essentially turns off the hotend. Is that correct?

Salandora, 2016-07-08:
yes, only bad effect of this command sequence is that it really waits until the hotend reaches 30 degree.

mabrown, 2016-07-08:
True, but it's at least worth considering. I usually shut the fan off at ~40C. Only time waiting would be an issue would we if I have another print job I want to do immediately. And it generally doesn't take that long to cool to 40.
Between the automated solution (which I'm thrilled to have -- been wanting one since I got the printer) and the Navbar Temp plugin, I'm happy to close this request, unless you guys would prefer it be left open.

ntoff, 2016-07-10:
Isn't it M109 R30 to wait for cooldown? I believe S only waits for heatup. If you're using repetier firmware it's even worse, it doesn't support R variables so you have to use M116 to make it wait for the tool to hit that temperature on cooldown.

http://reprap.org/wiki/G-code#M109:_Set_Extruder_Temperature_and_Wait

http://reprap.org/wiki/G-code#M116:_Wait

foosel, 2016-07-16:
You definitely don't want it to wait for 0degC (which it will never reach, if it even accepts that), so rather what @ntoff said, with the slight caveat that you want to added M116 to the list of long running commands in settings > serial for that too.

Otherwise this is something I'd suggested as a plugin anyhow, not as something to be implemented in OctoPrint directly, so if OP is happy with the navbar temp plugin already I'd say let's close this.

mabrown, 2016-07-16:
Gina, I'm happy with the plugin and have actually used the command sequence Salandora provided (modified to go to 40C rather than 30C) to automate the process anyway, so I'm closing the request. Thanks to everyone who contributed!

closed 2016-07-16 by mabrown