OctoPrint/OctoPrint #1405: Error while installing OctoPrint in Windows 10

Dreyer1416, 2016-07-10:
#### What were you doing?

Installing OctoPrint in WIndows 10, following the instructions in this link https://github.com/foosel/OctoPrint/wiki/Setup-on-Windows. Everything was OK until step 5.

#### What did you expect to happen?

To successfully Install Octoprint 

#### What happened instead?

This error message showed up in the command prompt :

```
Installed c:\octoprint\venv\lib\site-packages\octoprint-1.2.13-py2.7.egg
Processing dependencies for OctoPrint==1.2.13
Searching for regex
Reading https://pypi.python.org/simple/regex/
Best match: regex 2016.6.24
Downloading https://pypi.python.org/packages/96/4a/9943cf2bc0c60e6c502c1df407393696f97ac0209794ee827cf263cbcb60/regex-2016.06.24.tar.gz#md5=e75fcd602b73483543444e443faa4119
Processing regex-2016.06.24.tar.gz
Writing c:\users\vander\appdata\local\temp\easy_install-gkl61g\regex-2016.06.24\setup.cfg
Running regex-2016.06.24\setup.py -q bdist_egg --dist-dir c:\users\vander\appdata\local\temp\easy_install-gkl61g\regex-2016.06.24\egg-dist-tmp-pd70tf
C:\Octoprint\venv\lib\site-packages\setuptools\dist.py:285: UserWarning: Normalizing '2016.06.24' to '2016.6.24'
  normalized_version,
error: Setup script exited with error: [Error 2] The system cannot find the file specified
```

foosel, 2016-07-12:
Apparently `python setup.py install` is refusing to pull in the binary wheel of the `regex` dependency and instead uses the source archive which needs to be compiled before it becomes usable. My guess is you cannot compile the package (stock Win10 doesn't come with a properly setup python compilation environment) and hence the install fails.

It appears to do the proper thing however if - instead of `python setup.py install` - you perform a 

    pip install .

@Dreyer1416, can you test if that solves the problem for you as well? If so I'll adjust the setup guide.

Dreyer1416, 2016-07-12:
When I input `pip install` in the terminal I receive the following message 

`You must give at least one requirement to install (see "pip help install")`

foosel, 2016-07-12:
Try `pip install .` in the OctoPrint folder (so at the same location where you'd usually execute `python setup.py install`). The dot at the end is crucial and was not a typo above :)

Dreyer1416, 2016-07-12:
I only noticed the dot after I already posted the message... I already have deleted it

I typed `pip install .` in the terminal and a few errors appeared... Here is the full terminal output:
(Sorry for the wall of text but I have no idea which parts are important)

```
(venv) C:\Octoprint>pip install .
Processing c:\octoprint
Collecting flask<0.11,>=0.9 (from OctoPrint==1.2.13)
  Downloading Flask-0.10.1.tar.gz (544kB)
    100% |################################| 552kB 559kB/s
Collecting werkzeug==0.8.3 (from OctoPrint==1.2.13)
  Downloading Werkzeug-0.8.3.tar.gz (1.1MB)
    100% |################################| 1.1MB 480kB/s
Collecting tornado==4.0.1 (from OctoPrint==1.2.13)
  Downloading tornado-4.0.1.tar.gz (314kB)
    100% |################################| 317kB 1.2MB/s
Collecting sockjs-tornado==1.0.1 (from OctoPrint==1.2.13)
  Downloading sockjs-tornado-1.0.1.tar.gz
Collecting PyYAML==3.10 (from OctoPrint==1.2.13)
  Downloading PyYAML-3.10.tar.gz (241kB)
    100% |################################| 245kB 1.2MB/s
Collecting Flask-Login==0.2.2 (from OctoPrint==1.2.13)
  Downloading Flask-Login-0.2.2.tar.gz
Collecting Flask-Principal==0.3.5 (from OctoPrint==1.2.13)
  Downloading Flask-Principal-0.3.5.tar.gz
Collecting Flask-Babel==0.9 (from OctoPrint==1.2.13)
  Downloading Flask-Babel-0.9.tar.gz
Collecting Flask-Assets==0.10 (from OctoPrint==1.2.13)
  Downloading Flask-Assets-0.10.tar.gz
Collecting Flask-Markdown==0.3 (from OctoPrint==1.2.13)
  Downloading Flask-Markdown-0.3.tar.gz (165kB)
    100% |################################| 174kB 875kB/s
Collecting pyserial==2.7 (from OctoPrint==1.2.13)
  Downloading pyserial-2.7.tar.gz (122kB)
    100% |################################| 122kB 1.3MB/s
Collecting netaddr==0.7.17 (from OctoPrint==1.2.13)
  Downloading netaddr-0.7.17-py2.py3-none-any.whl (1.5MB)
    100% |################################| 1.5MB 249kB/s
Collecting watchdog==0.8.3 (from OctoPrint==1.2.13)
  Downloading watchdog-0.8.3.tar.gz (83kB)
    100% |################################| 92kB 861kB/s
Collecting sarge==0.1.4 (from OctoPrint==1.2.13)
  Downloading sarge-0.1.4.tar.gz (50kB)
    100% |################################| 51kB 1.3MB/s
Collecting netifaces==0.10 (from OctoPrint==1.2.13)
  Downloading netifaces-0.10.0.tar.gz
Collecting pylru==1.0.9 (from OctoPrint==1.2.13)
  Downloading pylru-1.0.9.tar.gz
Collecting rsa==3.2 (from OctoPrint==1.2.13)
  Downloading rsa-3.2-py2.py3-none-any.whl (43kB)
    100% |################################| 51kB 624kB/s
Collecting pkginfo==1.2.1 (from OctoPrint==1.2.13)
  Downloading pkginfo-1.2.1.tar.gz
Collecting requests==2.7.0 (from OctoPrint==1.2.13)
  Downloading requests-2.7.0-py2.py3-none-any.whl (470kB)
    100% |################################| 471kB 328kB/s
Collecting semantic_version==2.4.2 (from OctoPrint==1.2.13)
  Downloading semantic_version-2.4.2.tar.gz
Collecting psutil==3.2.1 (from OctoPrint==1.2.13)
  Downloading psutil-3.2.1-cp27-none-win32.whl (88kB)
    100% |################################| 92kB 1.3MB/s
Collecting awesome-slugify<1.7,>=1.6.5 (from OctoPrint==1.2.13)
  Downloading awesome-slugify-1.6.5.tar.gz
Collecting feedparser<5.3,>=5.2.1 (from OctoPrint==1.2.13)
  Downloading feedparser-5.2.1.zip (1.2MB)
    100% |################################| 1.2MB 371kB/s
Collecting Jinja2>=2.4 (from flask<0.11,>=0.9->OctoPrint==1.2.13)
  Downloading Jinja2-2.8-py2.py3-none-any.whl (263kB)
    100% |################################| 266kB 906kB/s
[... 173 lines cut ...]
    creating build
    creating build\lib.win32-2.7
    creating build\lib.win32-2.7\markupsafe
    copying markupsafe\tests.py -> build\lib.win32-2.7\markupsafe
    copying markupsafe\_compat.py -> build\lib.win32-2.7\markupsafe
    copying markupsafe\_constants.py -> build\lib.win32-2.7\markupsafe
    copying markupsafe\_native.py -> build\lib.win32-2.7\markupsafe
    copying markupsafe\__init__.py -> build\lib.win32-2.7\markupsafe
    running egg_info
    writing MarkupSafe.egg-info\PKG-INFO
    writing top-level names to MarkupSafe.egg-info\top_level.txt
    writing dependency_links to MarkupSafe.egg-info\dependency_links.txt
    warning: manifest_maker: standard file '-c' not found

    reading manifest file 'MarkupSafe.egg-info\SOURCES.txt'
    reading manifest template 'MANIFEST.in'
    writing manifest file 'MarkupSafe.egg-info\SOURCES.txt'
    copying markupsafe\_speedups.c -> build\lib.win32-2.7\markupsafe
    running build_ext
    building 'markupsafe._speedups' extension
    error: [Error 2] The system cannot find the file specified

    ----------------------------------------
Command "c:\octoprint\venv\scripts\python.exe -u -c "import setuptools, tokenize;__file__='c:\\users\\vander\\appdata\\local\\temp\\pip-build-dyhdrn\\MarkupSafe\\setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record c:\users\vander\appdata\local\temp\pip-i2fuw9-record\install-record.txt --single-version-externally-managed --compile --install-headers c:\octoprint\venv\include\site\python2.7\MarkupSafe" failed with error code 1 in c:\users\vander\appdata\local\temp\pip-build-dyhdrn\MarkupSafe\
```

foosel, 2016-07-13:
Now that is odd. It's actually attempting to build the wheels instead of just fetching those on pypi.

What does `pip --version` output?

Dreyer1416, 2016-07-13:
`pip 8.1.2 from c:\python27\lib\site-packages (python 2.7)`

By the way, I already did a `pip install --upgrade pip` as is recommended by the wiki page.

foosel, 2016-07-14:
Ok... I can reproduce this kinda (I see that it's still compiling some dependencies, which I did not anticipate). So I guess we need to get compiling working after all. Could you try installing [Microsoft Visual C++ Compiler for Python 2.7](http://www.microsoft.com/en-us/download/details.aspx?id=44266) and then try again?

Sorry for the hassle, it's tricky to reproduce the problem on my own machines since they are fully setup for development and I need you to test the workarounds in a user kind of setup.

foosel, 2016-07-14:
Btw, if just installing the above does NOT solve the issue, try `pip install setuptools>=6.0` in the virtual environment. I'm not sure if that's really needed though (but again, can't test myself).

Dreyer1416, 2016-07-14:
I just installed the C++ compiler and did a `pip install .` It seems everything is running smoothly now. 

Thanks :)

[image]

foosel, 2016-07-15:
Perfect! Did you only need to install the compiler or also upgrade setuptools?

foosel, 2016-07-16:
Assuming no setuptools update was necessary. Setup guide is adjusted.

closed 2016-07-16 by foosel

Dreyer1416, 2016-07-17:
Yes. There was no need to update setuptools.

foosel, 2016-07-17:
Perfect, thanks!