OctoPrint/OctoPrint #1406: Files in "trash" show up in file browser after deletion.

JoshMcCullough, 2016-07-11:
Using OctoPi/OctoPrint v1.2.10, when I upload a file and then delete it, it still shows up in the file browser but with "trash-~1/files/" in the filename. See attached image.

#### What did you expect to happen?
Deleted files should not show up in the file browser.

#### What happened instead?
Deleted files show up in the file browser.

#### Branch & Commit or Version of OctoPrint

1.2.10

#### Printer model & used firmware incl. version

Robo R1 Plus, Marlin

#### Browser and Version of Browser, Operating System running Browser

Chrome 51, Windows 10

#### Link to octoprint.log

https://gist.github.com/JoshMcCullough/6bf40bba57bceec37a0e8b13666dde73

#### Screenshot(s) showing the problem:

[image]

foosel, 2016-07-13:
@JoshMcCullough How exactly did you delete the files? The files that are being shown in the screenshot are files from your printer's SD card, and your printer's firmware doesn't filter out trash bin folder that Windows created on there.

Deleting a file on the printer's SD *through OctoPrint* should actually delete it properly (at least as long as the firmware doesn't do weird stuff but actually just removes the file when instructed to do so through `M30`), however deleting it through the Windows Explorer while using the SD as a thumb drive usually first moves it to Windows' trash bin, and that is simply a specifically named folder within the root of the drive the file was deleted from.

So it would be important to know how you deleted those files.

PS: The bot complained because you didn't copy the full template (or just left the prefilled template). Manually marking the ticket as complete.

JoshMcCullough, 2016-07-13:
Sorry, I should have indicated that I'm deleting the file via OctoPrint's delete button in the file browser. I'm pretty sure if I delete file `X.gcode` it will then show up as `/trash-~1/files/X.gcode`. I will confirm this tonight.

JoshMcCullough, 2016-07-15:
You're right, the trashed files were on the SD card! Thanks for pointing that out.

closed 2016-07-15 by JoshMcCullough