OctoPrint/OctoPrint #1410: [Request] Printrbot Plus Metal Support

Pghpunkid, 2016-07-15:
I have a newer Printrbot Plus printer, and was excited to use OctoPrint with it to clean up my office, but its not supported. Can we work on that? I am open to help anyway I can. 

Thanks!

foosel, 2016-07-16:
Was that the new one with custom firmware and custom apps to use?

Pghpunkid, 2016-07-16:
Its Model 1504. Im not aware of any custom apps or firmware. I have not modified in anyway..

Pghpunkid, 2016-07-16:
This document was found, however, there is no specific mention of Plus Models, and not sure if there is a major difference between Simple metal and Plus.

https://printrbot.com/wp-content/uploads/2014/04/RPi-Instructions.pdf

foosel, 2016-07-16:
Never mind, the model I meant is apparently not out after all (just remember reading something a couple months ago). 

All printrbots currently out should actually work just fine out of the box, what makes you come to the conclusion that it is unsupported? Did you try connecting to it from OctoPrint? What happened?

Pghpunkid, 2016-07-16:
The supported printers list on Octoprint's site says Printrbot Simple. I would imagine the printers profile would be different. I can try it if you believe it should work. I just don't want the printer to explode into a billion pieces. Dang thing cost $1300 US.

foosel, 2016-07-16:
The list on the wiki is not a case of "only what's on here will work" but rather "what's on here will definitely work and if your's isn't listed listed please test and update this list". 

And the worst thing that can happen in case it is actually unsupported (what I don't believe) is it simply not connecting at all. You also don't need any specific profiles just to connect to the printer in the first place, it's simply a serial connection.

Pghpunkid, 2016-07-16:
Glad to hear it. I will use that document I found on Printrbot's site, and give it a shot. Ill report back with information so it can be updated if it works. Thanks for the tip.

foosel, 2016-07-16:
The list is a wiki, the information in there is added by users just like you, so if it happens to work, please just add it yourself and close this ticket :)

Pghpunkid, 2016-07-16:
Will do. Thanks.

Pghpunkid, 2016-07-16:
Great news. After loading my Cura Profile and going through the basic configuration of the Printer Profile, it works amazingly. Thank you!

closed 2016-07-16 by Pghpunkid