OctoPrint/OctoPrint #1418: M3D restarting the print

bobbypop999, 2016-07-16:
I was just printing something and wanted to check how long it had left.

I expected the print to carry on normally 

Instead the printer stopped and restarted the print

I am using the current version.

I am using the M3D Micro.

I am using safari on an iPad.

I have read the FAQ.

foosel, 2016-07-16:
Please fill out the **full** ticket template. Currently there are no logs, no version number, no terminal output, no JavaScript console. That stuff is requested for a reason :)

donovan6000, 2016-07-17:
Make sure your using the latest version of [M33 Fio](https://github.com/donovan6000/M33-Fio), which at the time of writing this is V1.0. There's a bug in M3D Fio V0.35 and newer that can cause this to happen whenever a client connects to OctoPrint. Having both M3D Fio and M33 Fio installed at the same time can cause unexpected behavior, so it's recommended that you first uninstall M3D Fio then install M33 Fio.

bobbypop999, 2016-07-17:
I can't get the other information though.

bobbypop999, 2016-07-17:
Thank you it is now fixed.

closed 2016-07-17 by bobbypop999