OctoPrint/OctoPrint #1419: need help with talking to flashforge creator

mark-hahn, 2016-07-16:
When octopi http server tries to open a connection with my flashforge creator (makerbot replicator clone) it can't find the baudrate.  The terminal shows gibberish at the correct speed of 115200.  I suspect the printer is trying to talk binary, but that is a guess.  Octopi server has no problem talking to my printrbot using the same raspberry pi.

Has anyone had any luck talking to a makerbot or clone?  I saw several threads here talking about this in 2013 and I hope it has been resolved.

Thanks in advance ...

Salandora, 2016-07-16:
Hello,

FlashForge should become compatible with @markwal's plugin
http://plugins.octoprint.org/plugins/gpx/

markwal, 2016-07-17:
@mark-hahn, @Salandora is correct, the FlashForge is as you note a MakerBot Replicator (1) clone.  The MakerBot protocol is called x3g and is binary, not gcode ascii.  The GPX plugin translates between reprap style gcode and MakerBot style x3g protocol.  You need the plugin.

closed 2016-07-17 by markwal