OctoPrint/OctoPrint #1420: Trying to host webpage on my site

xSaktx, 2016-07-19:
I am trying to display the same OctoPrint webpage on my website. I am running a rpi with OctoPi. I cant really do port forwarding because I have no access to my router, is there another way I do this?

Salandora, 2016-07-20:
Hello @xSaktx,

First of all, this is a bug tracker ;-)
You have a question which might be better placed at the [G+Community:](https://plus.google.com/communities/102771308349328485741)

But to your question, it is problematically, I have no experience with this myself so just some rough guesses:
You could try to setup an vpn server which is accessible through the internet (e.g. on your website server?)
Next you connect with your OctoPi instance to this vpn server, this way you would have a connection between your octopi server and your website.
With some proxy (haproxy, nginx) you should be able to reroute the requests to your octopi server then.

I hope this helped a bit.

chasecrom, 2016-07-28:
Also, for what reason do you not have access to your router?

I faced a similar issue and just reset the router so that I could have the password.

closed 2016-08-03 by GitIssueBot

pcnate, 2016-08-04:
If you are running octoprint on an internal network and wish to display it on an external linux web server you could do a reverse ssh tunnel and then use the nginx web server to direct communication over that ssh tunnel. Such would allow you to enable SSL to encrypt your connection.

Here is a basic example: https://gist.github.com/gdamjan/4586758