OctoPrint/OctoPrint #1422: [Request] Add option for post roll for timed timelapse to duplicate last frame

CptanPanic, 2016-07-25:
I was trying to utilize the post-roll feature of the timelapse and realized that it does not really work as I would expect for timed timelapses.  I would like a 30 second post-roll but since I have my timer at every 10s, octoprint says it will keep recording for 30 seconds * 25fps * 10second interval = 125minutes.  But all I really want is to just have the last frame duplicated for the next 30 seconds like it does on the z-change timelapse.  Can we get an option for this?

mtowara, 2016-07-27:
I agree that this would be a useful addition and should maybe also be standard behavior.

foosel, 2016-11-22:
Merged to `devel`, will be part of 1.3.0rc2 (and naturally also 1.3.0)

foosel, 2016-11-25:
1.3.0rc2 is out

closed 2016-11-25 by foosel