OctoPrint/OctoPrint #1423: New update 1.2.14 creates connection problems

atc1441, 2016-07-28:
I updated the new version of octoprint today.

every time i try to start a print the connect gets lost. if i go back to the last version 1.2.10 everything works fine.

it is a delta printer and the printer don't get any Gcode after i start the print.

foosel, 2016-07-28:
Fill out the full ticket template please, as linked by the bot. There's simply no way to analyse this without the information requested therein.

jdcasey, 2016-07-28:
I'm not sure if this is the same issue the original reporter is running into, but I just updated to the latest OctoPrint. I had been running whatever version the updater recommended last time (I didn't check before updating, much to my chagrin), and ran a successful print not more than half an hour ago. After updating and trying a LOT of variations of rebooting the system, restarting the Arduino, rebooting everything, I'm stuck. The error I'm seeing in the terminal seems like it might be the cause:

```
Recv: �����is not Last Line Number+1, Last Line: 2
Recv: Resend: 3
Recv: ok
Send: N3 T0*57
Recv: echo:Unknown command: ""
Recv: ok
Send: N4 M190 S70.0*69
Recv: Z�Y.J��not Last Line Number+1, Last Line: 2
Recv: Resend: 3
Recv: ok
Send: N3 T0*57
Recv: echo:Unknown command: ""
Recv: ok
Send: N4 M190 S70.0*69
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N4 M190 S70.0*69
Recv: Errorg��not Last Line Number+1, Last Line: 2
Recv: Resend: 3
Recv: ok
Send: N3 T0*57
Recv: echo:Unknown command: ""
Recv: ok
Send: N4 M190 S70.0*69
Recv: [���is not Last Line Number+1, Last Line: 2
Recv: Resend: 3
Recv: ok
Send: N3 T0*57
Recv: echo:Unknown command: ""
Recv: ok
Send: N4 M190 S70.0*69
Recv: Erro�I����Last Line Number+1, Last Line: 2
Recv: Resend: 3
Recv: ok
Send: N3 T0*57
Recv: echo:Unknown command: ""
Recv: ok
Send: N4 M190 S70.0*69
Recv: �umber is not Last Line Number+1, Last Line: 2
Recv: Resend: 3
Recv: ok
Send: N3 T0*57
Recv: echo:Unknown command: ""
Recv: ok
Send: N4 M190 S70.0*69
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N4 M190 S70.0*69
Recv: ErBr is not Last Line Number+1, Last Line: 2
Recv: Resend: 3
Recv: ok
Send: N3 T0*57
Recv: echo:Unknown command: ""
Recv: ok
Send: N4 M190 S70.0*69
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N4 M190 S70.0*69
Recv: Error:L5r���Last Line Number+1, Last Line: 2
Recv: Resend: 3
Recv: ok
Send: N3 T0*57
Recv: echo:Unknown command: ""
Recv: ok
Send: N4 M190 S70.0*69
Communication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N4 M190 S70.0*69
Recv: �������is not Last Line Number+1, Last Line: 2
Recv: Resend: 3
Recv: ok
[... 209 lines cut ...]
Recv: �g��not Last Line Number+1, Last Line: 2
Recv: Resend: 3
Recv: ok
Send: N3 T0*57
Recv: echo:Unknown command: ""
Recv: ok
Send: N4 M190 S70.0*69
Recv: �umber is not Last Line Number+1, Last Line: 2
Recv: Resend: 3
Recv: ok
Send: N3 T0*57
Recv: echo:Unknown command: ""
Recv: ok
Send: N4 M190 S70.0*69
Recv: }[���is not Last Line Number+1, Last Line: 2
Recv: Resend: 3
Recv: ok
Send: N3 T0*57
Recv: echo:Unknown command: ""
Recv: ok
Send: N4 M190 S70.0*69
Changing monitoring state from 'Printing' to 'Operational'
Changing monitoring state from 'Operational' to 'Offline'
Connection closed, closing down monitor
```

jdcasey, 2016-07-28:
BTW, here's the gcode (attached)
[coin-cell-holder-24.5mm.gcode.txt](https://github.com/foosel/OctoPrint/files/389281/coin-cell-holder-24.5mm.gcode.txt)

KevMags, 2016-07-28:
Same here.  Heats bead then stalls, eventually times out.  Too bad that I updated 3 Pis before doing a test print.

jdcasey, 2016-07-28:
@foosel Hopefully my error report above is enough to remove the "incomplete ticket" label?

foosel, 2016-07-28:
Your firmware is sending heaps of garbage. Or at the very least OctoPrint is receiving heaps of garbage. If suggest [to roll back to 1.2.13](https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-revert-to-an-older-version-of-the-octoprint-installation-on-my-octopi-image) to see if that fixes it. That would get I'd another data point. On first glance this does look some serious noise on the line though.

Instructions for a roll back on OctoPi:

```
cd ~/OctoPrint
source ~/oprint/bin/activate
git reset --hard 1.2.13
python setup.py clean
python setup.py install
sudo service octoprint restart
```

To find out your former version, look into octoprint.log - the full version number gets logged on every start up of the server there (another reason why everyone should always provide a fully filled out ticket template, even when just posting a "me too" ;))

For the record, I've been printing one test print after the other over the past week with what is now 1.2.14. So whatever it is (if it is the update), it is something that depends on the environment as well. Exact firmware, settings etc could all be a factor here. Please provide more data :)

foosel, 2016-07-28:
OK guys, I need more data points here from *all* of you. What firmwares are these, how's the printer connected (usb or direct connection). And terminal output and octoprint.log from everyone please.

@jdcasey usually it is the original poster who's supposed to provide that, but yes

foosel, 2016-07-28:
Oh, and also the electronics used please. There must be some common factor here.

jdcasey, 2016-07-28:
After rolling back to 1.2.13, here's my terminal as I start that print I was trying: https://gist.github.com/jdcasey/80cdde193537244d6e4327f13f778359

Oh, and I realized I still have the old octoprint.log from 1.2.14 in my download cache: 
[octoprint.log.txt](https://github.com/foosel/OctoPrint/files/389343/octoprint.log.txt)

The octoprint log from after the rollback: 
[octoprint.1.2.13.log.txt](https://github.com/foosel/OctoPrint/files/389344/octoprint.1.2.13.log.txt)

As far as firmware, I'm not sure how to get at that information (I'm still learning the ropes here), but in the terminal log I reported above, it had:

```
FIRMWARE_NAME:Marlin V1; Sprinter/grbl mashup for gen6 FIRMWARE_URL:http://www.mendel-parts.com PROTOCOL_VERSION:1.0 MACHINE_TYPE:REPRAPGURU EXTRUDER_COUNT:1
```
I am using a USB connection from a RPi3B+ to an Arduino Mega. The RPi is using the OctoPi image, but was previously updated to a newer version of Octoprint after flashing the image IIRC. 

I'm not sure what other electronics you're interested in...

KevMags, 2016-07-28:
Ramps 1.4 running Marlin RC7.  Direct serial connection to Pi3.

Version 1.2.14 (master branch)

[serial.log.TXT](https://github.com/foosel/OctoPrint/files/389354/serial.log.TXT)

[octoprint.log.txt](https://github.com/foosel/OctoPrint/files/389358/octoprint.log.txt)

foosel, 2016-07-28:
@KevMags can you also provide a `serial.log` with an active print? I only see a connect and subsequent temperature queries in the one you posted :/

@jdcasey hm, nothing unusual in there that immediately jumps to my eye... I was talking about the printer electronics. RAMPS, Rumba, whatever it is that hosts your stepper drivers and is on the other end of the cable you plugged into the Pi.

KevMags, 2016-07-28:
Rolled back to 1.2.13, worked fine.  Updated again. Erased all logs.  Connected. Clicked icon "load and print".  Unhandled Com error.  Reconnected, clicked on "load file" icon.  Clicked on "print" button.
Unhandled Com error, bed still maintaining 60C.

Reconnected, restarted OctoPrint to write log files.

[serial.log.txt](https://github.com/foosel/OctoPrint/files/389423/serial.log.txt)
[octoprint.log.txt](https://github.com/foosel/OctoPrint/files/389424/octoprint.log.txt)

foosel, 2016-07-28:
What's written to the Terminal tab when that "Unhandled Com Error" happens?

foosel, 2016-07-28:
FYI, I've put further roll-outs of 1.2.14 on hold until we've gotten to the bottom of this.

What needs to be figured out now is a) if all of you really are seeing identical issues and b) what is causing this issue/these issues so that c) I can reproduce them (I can't *at all* right now) and fix it/them.

foosel, 2016-07-28:
Copied over from #1426:

> So here is a clip from the terminal after I preheat the heated bed, and extruder myself.
> 
> ```
> ecv: T:200.13 E:0 B:69.8
> Recv: T:200.00 E:0 B:69.9
> Recv: T:200.04 E:0 B:69.8
> Recv: T:199.93 E:0 B:69.8
> Recv: T:200.00 E:0 B:69.8
> Recv: T:199.93 E:0 B:69.9
> Recv: ok
> Send: N6 M104 S200 T037
> Recv: echo:Unknown command: ""
> Recv: ok
> Send: N7 M109 S200 T041
> Recv: Error:Line�st Line Number+1, Last Line: 5
> Recv: Resend: 6
> Recv: ok
> Send: N6 M104 S200 T037
> Recv: echo:Unknown command: ""
> Recv: ok
> Send: N7 M109 S200 T041
> Recv: Err� ����Last Line Number+1, Last Line: 5
> Recv: Resend: 6
> Recv: ok
> Send: N6 M104 S200 T037
> Recv: echo:Unknown command: ""
> Recv: ok
> Send: N7 M109 S200 T041
> ommunication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication > timeout if that happens regularly on specific commands or long moves.
> Send: N7 M109 S200 T0*41
> Recv: Error:Line Num��r������1, Last Line: 5
> See octoprint.log for details
> Changing monitoring state from 'Printing' to 'Offline: See octoprint.log for details'
> Connection closed, closing down monitor
> ```
> 
> so thats where I am, it's happened 3x I even rebooted the Pi just to make sure.

KevMags, 2016-07-28:
Heats bed, prints "Bed Done" on LCD.
Seems to choke after:  Send: N2 M109 S210.000000*71

Text cut and paste from terminal tab:
[TerminalTab.txt](https://github.com/foosel/OctoPrint/files/389460/TerminalTab.txt)

foosel, 2016-07-28:
@KevMags try setting these to 0 please, save, reconnect:

[image]

Not sure if that will solve the problem or just possibly remove something masking the real issue.

KevMags, 2016-07-28:
Having to cold boot ramps for it to accept any instructions after having trouble connecting (after setting timeouts to 0).  It takes roughly 1-2 minutes after "Send: N2 M109 S210.000000*71" before it throws the error (before setting timeouts to 0).  After setting timeouts to 0 it took....... Well it seems stuck now for about 10 minutes with no error and the nozzle never heats.

After about 10 minutes I restarted the Pi to force log write.

Aaaak! That printer is not responding to bed heating anymore.  Moving to another printer.

Printer #2.  Bed heats, OctoPrint sends:"Send: N2 M109 S210.000000*71".  No nozzle heating, no errors after 5 minutes (timed).
Restart OctoPrint for force log file write.

Massive connection problems, unresponsive to bed heat commands.
Here we go, it is heating now.  Aaaaand now we have some errors.
Annnnnd it is printing.

Here are the new files from different printer:
[octoprint.log.txt](https://github.com/foosel/OctoPrint/files/389569/octoprint.log.txt)
[Printer2term.txt](https://github.com/foosel/OctoPrint/files/389571/Printer2term.txt)

Serial log is empty.
Will try again and report back.

KevMags, 2016-07-28:
Printer #2 printed.  After cancelling print, it no longer heats bed.  Warm reboot Pi, no longer heats bead.  Reset ramps with reset switch, bed heats, nozzle heats, and prints.

foosel, 2016-07-28:
@KevMags so you are rather seeing a complete stop in responses from the printer for a time, but not the garbled response lines as observed by @jdcasey and @sLpFhaWK, correct? Looks like two symptoms then, probably of the same issue considering that they vanish when rolling back, but manifesting differently for you. 

Is anyone (or all) of you familiar with git bisect? That plus a bit of testing would help pinpointing the exact commit that introduced the problem. I stared at the code for the past two hours and tried a bunch of ideas to reproduce the same issues, but so far I'm drawing a complete blank, I simply cannot reproduce any of the observed behaviours at all. Knowing at least which exact commit introduced the problem might help getting an idea what the issue is (and how to solve it).

In any case, I'll have to stop for today now, it's 1:30am here and it's getting hard to keep my eyes open, despite me desperately wanting to figure this out.

KevMags, 2016-07-28:
I see garble responses at times.  I am running 0 for timeouts now.

Printer #3
Heats bed and stabilizes at 60C, never prints "bed done" to LCD.  Hangs after " N5 M190 S60*91"

Printer #3 Files:
[Prntr3Term.txt](https://github.com/foosel/OctoPrint/files/389608/Prntr3Term.txt)
[octoprint.log.txt](https://github.com/foosel/OctoPrint/files/389609/octoprint.log.txt)

KevMags, 2016-07-28:
Printer #1 will connect, shows proper temps but not respond to "print" or to manually set bed temp.  No response in terminal window after "Send: N1 M190 S60.000000*113"

jdcasey, 2016-07-29:
Okay, I bisected all the way through between the last release commit and the current one. I never hit a problem. When I got back to the 1.2.14 release commit, I decided to try something, and I rebuilt that using the setup instructions you gave for rolling back. It's in the other room right now, printing a part for me. It's running 1.2.14.

My current theories are that either the automatic update feature involves downloading some pre-compiled code that's incompatible with the flavor of ARM that the RPi3 uses, or else it didn't do a full clean of the current environment, and wound up with parts of two versions trying to function together. 

My gut tells me, from the garbled terminal messages, that there's something about the CPU architecture that messed with the update...maybe endian-ness? That's a wild-ass guess, obvs. It's a little hard to see how an outdated python binary could lead to some parts of the comms being garbled, but others being fine. For that matter, something as fundamental as endian orientation should cause everything to be garbled too, I would think, like setting the wrong baud rate.

Ok, enough conjecture from me. I just wanted to say that I have a RPi3 and when I *built* the 1.2.14 release it worked fine.

KevMags, 2016-07-29:
Please help a python illiterate old man out here.   Are you saying that I can build a 1.2.14 version by using the following and replacing 1.2.13 with 1.2.14?
 
cd ~/OctoPrint
source ~/oprint/bin/activate
git reset --hard 1.2.13
python setup.py clean
python setup.py install
sudo service octoprint restart

jdcasey, 2016-07-29:
@KevMags Yep, that's effectively what I did. Oops...I just realized I didn't use the second command. I had some trouble when I started, and had to install setuptools using apt-get. I think that second command sets up a localized python environment to avoid the need for all that.

I think the commands you listed should do the trick. If not, that's a useful data point, since mine did (but I messed up the environment settings, which is what the second command is).

jdcasey, 2016-07-29:
Okay, I rebuilt 1.2.14 using the activate script to set the environment, and it froze up again. I'm going to go back through the bisect using the right procedure this time and see where it breaks.

jdcasey, 2016-07-29:
TL;DR: It appears that the problem came in at commit 937487037aeae452d1656e2fc38870b8203eb5a6.

So, building using the global environment works, and that's interesting IMO. However, using the localized python virtualenv from ~/oprint/bin/activate, the git bisect took a lot longer to home in.

Here are my findings: https://gist.github.com/jdcasey/bee6b435d70f50b9efada6e7a919c7a5

foosel, 2016-07-29:
@jdcasey thank you **SO MUCH** for this. I would have never guessed it could be this commit, I was looking at the changes I did to sending.

Would you be available for testing a "1.2.14 + reverted 9374870" branch in a couple minutes?

jdcasey, 2016-07-29:
@foosel I probably have about one more test in me tonight. It's about 1am here, and I need to get a few hours before work tomorrow... :(

foosel, 2016-07-29:
@jdcasey sounds familiar, big thanks for the help so far! Without being able to reproduce this myself I really am completely hamstrung.

Ok, could everyone who can reproduce these problems please try if the `fix/revertTimeoutAdjustment` branch solves the problem for them? That removes the re-setting of the read timeout on the serial to a shorter value during a heatup.

----

To give this a whirl on OctoPi:

```
cd ~/OctoPrint
source ~/oprint/bin/activate
git fetch
git checkout fix/revertTimeoutAdjustment
python setup.py clean
python setup.py install
sudo service octoprint restart
```

----

Instructions for manual install which followed the setup guide:

```
cd /path/to/OctoPrint
source venv/bin/activate
git fetch
git checkout fix/revertTimeoutAdjustment
python setup.py clean
python setup.py install
```

+ restart OctoPrint

jdcasey, 2016-07-29:
@foosel I tried the top set of instructions above. It's printing now. I'd say whatever you changed, you got it.

foosel, 2016-07-29:
That's good news I guess, although I'm *completely* baffled by this and don't understand how setting the timeout (which is basically all that I removed just now) can have these consequences O_O Not understanding why code creates an issue severely worries me because that means I can't avoid the core cause in the future.

@atc1441 @KevMags @sLpFhaWK could you give this a test as well?

jdcasey, 2016-07-29:
@foosel Speaking as a python newbie, the only thing I can find in that latest commit that stands out is:

https://github.com/foosel/OctoPrint/commit/1f19fd1e1db5edc1bfacd8e9ef57926721f0be95#diff-493d6fa72f9017192b4aaf4a824e2e0aL1358

There are two calls assigning values to `self._serial.timeout` that aren't in the reverted code. Everything else looks that same. Is it possible that that value cannot be changed in an open connection? Like I say, just guessing...

Ok, really time for bed here. 😄

foosel, 2016-07-29:
@jdcasey It can, and it actually does it elsewhere in the code since a couple of years (e.g. during connection handshake). Plus it didn't cause any problems on my own printer here and I've been printing a lot over the past week to test the changes I did to the comm layer. So it's not a general issue, but it appears to be causing trouble for some people/machines.

I'm digging through my test hardware now to see if I can find anything with which I can reproduce this myself, to be able to better understand it. I'm also still worried a bit that the issue does manifest in separate ways and not 100% sure that there might not be another problem hidden in there.

Anyhow, a good night to you and big thanks again for the `git bisect`, that really made the difference between frustrated head scratching here and at least an idea where to dig to understand this (while of course also preparing a fix release once I'm sure it really only is this issue).

foosel, 2016-07-29:
Good news. I could reproduce it on the UM I still had collecting dust in a cupboard. For me it shows the behaviour that @KevMags described, not the garbled messages observed by @jdcasey, @sLpFhaWK and also some people on the G+ community, and the fix instantly solves it.

That still doesn't explain what is going wrong there in the first place but at least it looks like pushing that fix out would solve it. Still, the more reports I can get to confirm this, the better, so please, test.

foosel, 2016-07-29:
For reference, minimal test file: https://gist.github.com/foosel/2264d61a979833c616717095079742d9

foosel, 2016-07-29:
The fix is now merged to `maintenance`, which also contains a fix for an issue with certain disconnect constellations as observed by @stema12 in #1425 and a compatibility fix for plugins which replace the serial object and don't return the amount of written bytes on their `write` implementation.

New instructions for testing as follows, please test and report back - I don't want to release this before I'm not sure it's really fixing the issue for everyone who so far chimed in here.

----

To give this a whirl **on OctoPi**:

```
cd ~/OctoPrint
source ~/oprint/bin/activate
git fetch
git checkout maintenance
python setup.py clean
python setup.py install
sudo service octoprint restart
```

----

Instructions for **manual install** which followed the setup guide:

```
cd /path/to/OctoPrint
source venv/bin/activate
git fetch
git checkout maintenance
python setup.py clean
python setup.py install
```

+ restart OctoPrint

sLpFhaWK, 2016-07-29:
I will test but I won't be able to do it until later today after I. get home from work. I'm just getting ready now. glad to see it possibly resolved though!!

stimpy23, 2016-07-29:
Sorry for my super-incomplete bug reporting, but I hade similar errors after updating to 1.2.14 on my FlashForge Creator Pro with Sailfish v7.7 and testing your maintenance release solved 'em for me, so did reverting to 1.2.13 afterwards..
You guys are awesome! Was too lazy filing a complete bugreport yesterday late at night and today, I was able to read the whole story including a simple solution over here.. Thank you very much!

foosel, 2016-07-29:
Everyone keep testing please. I'm not completely convinced that we got it after all. I was just going through my release checklist, this time also doing several previous/after tests against the UM and while I couldn't reproduce at all against 1.2.15.dev I could also sometimes not reproduce it against 1.2.14 and that worries me that there might still be something else in there.

I'm currently leaning towards leaving the 1.2.14 release disabled but not pushing 1.2.15 until after the weekend when hopefully more people have had a chance to test this thoroughly. I'll need your cooperation there however, because I only have one machine here that I can reproduce the problem semi reliably on and that is not the base of judgement I'd like here.

KevMags, 2016-07-29:
Fixed all 3 of my printers.  Thanks!!!

foosel, 2016-07-29:
@KevMags Glad to hear, thanks for reporting back!

In the meantime I've dug a bit deeper still as to why I suddenly couldn't reproduce it reliably anymore against the same printer I could reproduce it just fine this very morning. Unreliable reproduction was on my test Pi2. Reliable reproduction on my development machine. So that got me thinking. Forcing a bit of a wait between sending the `M109` and the consequent read timeout setting made it magically work on my workstation too. Reliably. Every single time. So my current guess is that there's something inside pyserial or possibly the underlying OS code that's responsible for this. Changing the read timeout constitutes as reconfiguring the port and my guess is that's bound to create trouble if there are still bytes being sent at another corner of the system via said port. Adding a little delay before changing the timeout gave the machine enough time to send all bytes over the line and empty its buffers so that the timeout switch didn't cause any harm.

I'm still not fully sure why there's a difference in behaviour between my regular to-go printer (no issue reproducible at all) and the UM I finally could trigger the problem on, but my guess would be either different buffer settings or chipsets or something along those lines.

At least, the current level of understanding of what's happening here makes me a bit more calm about pushing the fix as an actual fix. Still, I'll wait a couple more hours at a minimum, or possibly over the weekend, depending on how much feedback I get in the meantime. I really don't want to have to push the third release within 48 hours ;)

jdcasey, 2016-07-29:
@foosel Okay, I updated to the maintenance branch and reinstalled. It's currently printing. Started without a hitch, and the terminal output looks clean.

foosel, 2016-07-29:
@jdcasey great news then I guess! :) 

That makes a 100% success rate so far over the posts on the G+ community and everyone who reported back here. And considering that I now have an idea about what went wrong I feel better about releasing this.

I'll still leave you people on the other side of the Atlantic a bit more testing time, so that @sLpFhaWK and @atc1441 also have a chance to join the party... I need a full night's sleep anyhow (hopefully without dreaming about OctoPrint's communication layer this time ;)) and hence will wait with the release until tomorrow morning.

Big thanks to everyone for reporting back their findings, and especially to @jdcasey for jumping through the hoops of the git bisect that uncovered the guilty commit!

atc1441, 2016-07-29:
@foosel
Thank you for your sleepless nights and for the hard but good work on octoprint.
I didnt had the time to do a bigger bug search, sry for that.

Have a nice day or night

jdcasey, 2016-07-29:
@foosel I'm happy to help. It was fun debugging a software problem that could cause a fire. :smile: Most of my stuff is so boring by comparison.

sLpFhaWK, 2016-07-29:
I am headed home now and when I get there I will apply the fix you have listed! 30-45 mins till I report back! thanks!

sLpFhaWK, 2016-07-29:
I am happy to report that the fix worked for me, Printer heated up normally and started to print just like before. I'm glad it works, thanks everyone for the help!

JustSid, 2016-07-29:
I'd also like to chime in, it works great for me again from the maintenance branch. Thank you so much for the quick fix @foosel!

BinaryConstruct, 2016-07-30:
From my testing the maintenance branch also fixes the comms issues I was having with 1.2.14.

Hardware:
RPi 3
Sanguinololu with Marlin RC_bugfix https://github.com/MarlinFirmware/Marlin/commits/RCBugFix
Ramps + Arduino Mega with Marlin RC_Bugfix

foosel, 2016-07-30:
Thanks everyone! 1.2.15 release with the fix is now published. I for one wish everyone a nice weekend, I know I really need mine now ;)

closed 2016-07-30 by foosel