OctoPrint/OctoPrint #1424: [Brainstorming/Question] How does OctoPrint populate the temperature graph?

chasecrom, 2016-07-28:
I'm looking into ways to improve OctoPrint, but I don't quite understand how OctoPrint gets the temperature data. Could someone point me to the right files?

FirefighterBlu3, 2016-09-12:
octoprint sends M105 commands when idle and in between every few commands when printing. M105 instructs the printer to emit the temperatures of the extruder(s) and build base (heated bed).

for example:
```
Send: N1197 M105*25
Recv: ok T:183.9 /185.0 B:59.3 /60.0 @:69 B@:0
```

here, octoprint is in the middle of one of my print jobs. N1197 is the current line number that is being printed, M105 is the request for temperatures, *25 is the checksum of the characters starting from N and ending at the 5 before the asterisk.

my printer responds with the temperatures, T:183.9 is my extruder actual temperature in Celsius, followed by 185.0 which is the set temperature. next is the build plate which is currently at 59.3C and is set to 60.0. (i don't know what the @: components are)

foosel, 2016-11-26:
Considering this answered ;)

closed 2016-11-26 by foosel