OctoPrint/OctoPrint #1426: Same issue with others, connectivity issues.

sLpFhaWK, 2016-07-28:
So here is a clip from the terminal after I preheat the heated bed, and extruder myself.

ecv: T:200.13 E:0 B:69.8
Recv: T:200.00 E:0 B:69.9
Recv: T:200.04 E:0 B:69.8
Recv: T:199.93 E:0 B:69.8
Recv: T:200.00 E:0 B:69.8
Recv: T:199.93 E:0 B:69.9
Recv: ok
Send: N6 M104 S200 T0*37
Recv: echo:Unknown command: ""
Recv: ok
Send: N7 M109 S200 T0*41
Recv: Error:Linest Line Number+1, Last Line: 5
Recv: Resend: 6
Recv: ok
Send: N6 M104 S200 T0*37
Recv: echo:Unknown command: ""
Recv: ok
Send: N7 M109 S200 T0*41
Recv: Err�	����Last Line Number+1, Last Line: 5
Recv: Resend: 6
Recv: ok
Send: N6 M104 S200 T0*37
Recv: echo:Unknown command: ""
Recv: ok
Send: N7 M109 S200 T0*41
ommunication timeout during an active resend, resending same line again to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
Send: N7 M109 S200 T0*41
Recv: Error:Line Num��r������1, Last Line: 5
See octoprint.log for details
Changing monitoring state from 'Printing' to 'Offline: See octoprint.log for details'
Connection closed, closing down monitor

so thats where I am, it's happened 3x I even rebooted the Pi just to make sure.

foosel, 2016-07-28:
I moved this to #1423 - please only one ticket per issue, or I'll have to collect info from all over the place :)

closed 2016-07-28 by foosel

sLpFhaWK, 2016-07-28:
sorry first time ever using github for a ticket. I thought I was adding too it. sorry!

foosel, 2016-07-28:
No harm done!