OctoPrint/OctoPrint #1427: Timelapse Rendering Failed

TTEEDD, 2016-07-30:
Hello,
Every-time a print finishes it says this.
http://imgur.com/yQwPdSv
Please help me solve this issue i want to get the time-lapse feature running also i am using a Logitech USB Webcam

TTEEDD, 2016-07-31:
Okay so i managed to solve it some how but all i did was reinstall OctoPi and its working now.

foosel, 2016-08-15:
Closing then

closed 2016-08-15 by foosel

franzbischoff, 2017-08-15:
Hi, I'm having the same issue as @TTEEDD.

I'm using OctoPrint windows instalation and this is the error in terminal:

[image]

Popup message:

[image]

Happens the same after trying to re-render:

[image]

franzbischoff, 2017-08-15:
Fixed: I was using a not compatible ffmpeg instalation. Re-downloaded from website and its ok.

If you get "Output file #0 does not contain any stream" that's because the timelapse-0.jpg file is corrupt.

tis-cs, 2019-02-26:
Hi @franzbischoff 
i have the same issue. 
[image]

I can't find any information about compatible versions.
I have installed the newest version via apt:

```
ffmpeg version 3.2.12-1~deb9u1+rpt1 Copyright (c) 2000-2018 the FFmpeg developers
built with gcc 6.3.0 (Raspbian 6.3.0-18+rpi1+deb9u1) 20170516
configuration: --prefix=/usr --extra-version='1~deb9u1+rpt1' --toolchain=hardened --libdir=/usr/lib/arm-linux-gnueabihf --incdir=/usr/include/arm-linux-gnueabihf --enable-gpl --disable-stripping --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libebur128 --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmp3lame --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzmq --enable-libzvbi --enable-omx --enable-omx-rpi --enable-mmal --enable-openal --enable-opengl --enable-sdl2 --enable-libdc1394 --enable-libiec61883 --arch=armhf --enable-chromaprint --enable-frei0r --enable-libopencv --enable-libx264 --enable-shared
libavutil      55. 34.101 / 55. 34.101
libavcodec     57. 64.101 / 57. 64.101
libavformat    57. 56.101 / 57. 56.101
libavdevice    57.  1.100 / 57.  1.100
libavfilter     6. 65.100 /  6. 65.100
libavresample   3.  1.  0 /  3.  1.  0
libswscale      4.  2.100 /  4.  2.100
libswresample   2.  3.100 /  2.  3.100
libpostproc    54.  1.100 / 54.  1.100
```
Do you have a hint for me?

Thanks in advance!

franzbischoff, 2019-02-27:
Hi @tis-cs ,

I was using a windows installation at that time, do you have any log output?

tis-cs, 2019-02-27:
Hi @franzbischoff,

the octoprint.log says:

```
2019-02-27 09:43:56,343 - octoprint.timelapse - ERROR - Could not render movie due to unknown error
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/timelapse.py", line 842, in _render
    p = sarge.run(command_str, stdout=sarge.Capture(), stderr=sarge.Capture())
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1462, in run
    p.run(input=input, async_=async_)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1071, in run
    self.run_node(node, input=input, async_=False)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1187, in run_node
    result = getattr(self, method)(node, input, async_)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1333, in run_command_node
    node.cmd.run(input=input, async_=async_)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/fixes.py", line 24, in decorated_function
    return f(*args, **kwargs)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 662, in run
    self.stdin = p.stdin
UnboundLocalError: local variable 'p' referenced before assignment
```

tis-cs, 2019-02-27:
I found my fault. I put in "/usr/bin/ffmpeg/" as path to ffmpeg. The server says it is valid but it has to be "/usr/bin/ffmpeg" without "/" in the end.
 
correct config:

[image]

incorrect config, but shown as valid:

[image]