OctoPrint/OctoPrint #1429: Cost Estimator Plugin does not show available version in Software update

bmbeverst, 2016-07-31:
#### What were you doing?
I Installed the Cost Estimator Plugin. http://plugins.octoprint.org/plugins/cost/
I did it manually and with the plugin manager. The results are the same.

#### What did you expect to happen?
I expected the software update plugin to show the available version as being the same as the installed version.

#### What happened instead?
It should no version as being available. It is blank next to available.

#### Branch & Commit or Version of OctoPrint
1.2.15 stable

#### Browser and Version of Browser, Operating System running Browser
Chromium OS 
Version 50.0.2661.105 (64-bit)
Platform 50.1.99 (Developer Build - neverware) developer-build chromeover64

#### Screenshot(s) showing the problem:
[image]

I have read the FAQ.

foosel, 2016-08-01:
That's an issue with the cost estimator plugin's implementation of the software update information hook, not with OctoPrint. You need to report it there instead.

closed 2016-08-01 by foosel

bmbeverst, 2016-08-03:
OK, will do! Can you point me to any docs for how to set that up?

foosel, 2016-08-15:
Using the issue tracker here: https://github.com/jasiek/OctoPrint-Cost/issues

The plugin says to track Github releases but didn't actually create any Github releases, only tags.