OctoPrint/OctoPrint #1430: STL Viewer causes TypeError in octoprint.plugins.softwareupdate

bmbeverst, 2016-07-31:
#### What were you doing?
Installed the STL viewer plugin and checked for an update.

#### What did you expect to happen?
Expected to see the installed and available stl viewer versions.

#### What happened instead?
Instead the version is unknown and installed and available are blank.

#### Branch & Commit or Version of OctoPrint
1.2.15 stable

#### Browser and Version of Browser, Operating System running Browser
Chromium OS 
Version 50.0.2661.105 (64-bit)
Platform 50.1.99 (Developer Build - neverware) developer-build chromeover64

#### Link to octoprint.log

https://gist.github.com/bmbeverst/bda4986a92d58f1b5012097186616e92

#### Screenshot(s) showing the problem:
[image]

I have read the FAQ.

foosel, 2016-08-01:
That's an issue with the plugin's implementation of the software update information hook, not with OctoPrint. You need to report it there.

closed 2016-08-01 by foosel

jneilliii, 2016-08-03:
Gina, is there a change documented somewhere on the software update information hooks?  I don't think this was an issue until recent versions with my plugin.  I've looked at the changelog and didn't see anything specific.

foosel, 2016-08-15:
@jneilliii there was no change at all in how update checks are parsed for ages. Taking a quick look at your source your plugin doesn't seem to define any however, so I'm not sure why it's even showing up in the software update dialog in the above screenshot (it doesn't show up for me when giving it a quick test). Maybe another plugin that (accidentally) hijacked your identifier for the update configuration?