OctoPrint/OctoPrint #1431: [Feature request] Delayed print start

sanchosk, 2016-07-31:
I came with an idea for a feature.
This may be possible to be added by plugin only, unfortunately I am not skilled enough in Python :(
The idea is to program something like "delayed start" like you have on microwaves or dishwashers.
My printer makes a lot of noise, so running in over night is not an option.
I sliced and prepared a print now - at 2am - and it would be great if it started to print at 8am without the need to wake me up and hit start on my cellphone :)

timmit99, 2016-08-01:
You could look into creating a custom command so you can start your print and then use this button to send a delay for a certain number of hours
https://github.com/foosel/OctoPrint/wiki/Cookbook:-Custom-Controls

This command MIGHT work as is. I'm not at my printer right now to test it.
I'm trying to thin of a way that you could have this automatically add this as the FIRST line in any print from octoprint.

- name: Delay Start
  type: parametric_command
  command: G4 P%(delay*3600000)p
  input:
    - name: Delay in Hours (0-24)
      parameter: delay
      default: 0

sanchosk, 2016-12-28:
Just to tell everyone interested - I made this using REST API call.
I use linux "at" command to call a simple script that will load the last file and start printing :)

closed 2016-12-28 by sanchosk