OctoPrint/OctoPrint #1432: Filament usage estimates are wrong when using volumetric extrusion

mikenew12, 2016-08-01:
#### What were you doing?

Uploaded a gcode file to OctoPrint that uses volumetric extrusion (generated from Slic3r in this case, with `Printer->Use Volumetric E` turned on). The diameter of the filament is set to 1.75 in the printer firmware itself with a `M200 D1.75` command.

#### What did you expect to happen?

I'd expect (or hope at least) that OctoPrint would be able to estimate the filament usage. I don't know that there would be a way to parse the gcode file and know that it's using volumetric extrusion, but if there was a checkbox or something in the printer profile to say "use volumetric extrusion" that would be more than fine. The filament diameter is already entered, so the calculation would be trivial.

#### What happened instead?

OctoPrint shows filament usage numbers that are far too big. An object that actually uses 293.5cm³ of filament will show as using 706.00cm³.

Converting is really simple. Since E is being defined in mm³ rather than in mm extruded, you'd simply use a formula like this:
`current_estimate / ((filament_diameter × .5)^2 × pi)`. 

So in the above example, using 1.75mm diameter filament would be:
`706.00cm³ / ((1.75mm * .5)^2 * 3.14) = 293.5cm³`

A good reference would be this (scroll to the bottom): http://reprap.org/wiki/Triffid_Hunter's_Calibration_Guide

#### Branch & Commit or Version of OctoPrint

Version: 1.2.13.post33+g58ee5f1 (HEAD branch)

#### Printer model & used firmware incl. version

Homebuilt Prusa-esque printer running Smoothieware commit 150b89e (edge branch).

#### Browser and Version of Browser, Operating System running Browser

Vivaldi, macOS 10.11.6

#### Link to octoprint.log

I can't image this matters. Just making the bot happy.

http://pastebin.com/wph3HGhn

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s) showing the problem:

I have read the FAQ.

CapnBry, 2016-08-12:
More important than the formula to convert the value is the question of how to determine that the estimate is a volume vs a length. There's no standard gcode that says "These are volumetric extrusion moves". You could look for the presence of a non-zero M200 Dxx, which indicates a filament diameter, but that's not guaranteed to be in the gcode and I would imagine most people don't have that because they set the diameter in the printer's menus itself (using the same gcode and adjusting filament diameter in the menu when switching spools).

It seems like the gcode analyzer needs to return a map of values instead of just a precomupted string. The extrusion distance, filament diameter, and nozzle diameters can be present if parsed from the gcode file and then leave it up to the client to render the number based on settings in the UI. This would allow more complex views of the data as well:

* Length of extrusion in any units (meters, feet, parsecs)
* Volume of extrusion (cm3, cc, l, or olympic swimming pools of plastic)
* Mass of plastic used (g, kg, lbs)
* Cost of plastic used (dollars, eurodollars, or Triganic Pu)

Then the UI needs to allow the user to enter defaults for anything unknown (not directly in the file): filament diameter, nozzle diameter, density of plastic used, cost per kg of plastic, which units you'd prefer, and a format string for what to display like "{mass}g / ${cost}". In short, this is a pretty big thing.

I have considered it but I just know that the length in meters that OctoPrint is displaying is actually volume if it is volumetric extrusion,, otherwise the volume is the actual volume.

mikenew12, 2016-08-12:
Well I looked into it a bit more. And you're right @CapnBry, the real question is how to figure out if E movements are intended as volume or distance.

After looking into it more, **Slic3r** actually makes it easy. At the bottom of the file there's `; use_volumetric_e = 1` whenever volumetric extrusion is turned on. So `gcodeInterpreter.py` could just look for that setting in the same place it's looking for `filament_diameter` and then adjust the calculations. Super simple.

**Cura**, however, doesn't have a setting for volumetric extrusion. To make it work you just set the filament diameter to 1.128379 ( sqrt(1/pi)*2 ) and it works out. So if OctoPrint doesn't adjust the calculations at all you actually get the correct volume calculation, but the length calculation is off. I can't think of any way to fix that without having some kind of toggle in the UI. Maybe the best option is to just ignore that and wait for Cura to implement it.

As a side note, OctoPrint isn't currently able to calculate the print volume from Cura-produced gcode, because Cura changed the name from `filament_diameter` to `material_diameter` in recent versions. Maybe that should be a separate bug report.

mikenew12, 2016-08-12:
@CapnBry I think the other issue you brought up (being able to set whatever units you like, calculate in mass vs. volume, etc.) would be really nice, but maybe should be treated separately. 

There's a couple plugins that estimate cost, filament usage, power, etc., and all of them are broken for me since they rely on inaccurate calculations. I think once that's resolved it might make sense to work on making the units more flexible, but that could also be done in a plugin.

CapnBry, 2016-08-19:
Yes it is true that some slicers include the data to indicate volumetric extrusion mode, but most do not. It would be adding code which could possibly support correct volumetric extrusion but only for Slic3r-origin gcode. Then you'd have a toggle for "cura mode length fix". It wouldn't *appear* to be any more accurate than the current numbers for the majority of people.

Even if you can tell from Slic3r's gcode that it is volumetric E, well ok now you've got the volume but not the length. There's no way to calculate length from volume if you don't know the filament diameter. You can assume that the diameter is what is baked into the gcode, but that's sort of the whole point-- that diameter won't be baked into the gcode because you intend to set it with M200.  I would say the only thing you really could do is look for "use_volumetric_e" and if it is then throw out the calculated volume and use the length as the volume (so length would be 0). Given that we won't know length, and volume is already shown by OctoPrint (labeled as length), then isn't that sufficient?

mikenew12, 2016-08-19:
Well, a few things:

1) I definitely don't think the way it works now is sufficient. Yes, you can just read the length as volume, but there are a handful of plugins designed to track filament usage, cost, etc., and they don't work. I print and ship nearly 100 parts a month, and not being able to easily track filament usage really sucks. Plus.. I mean it's wrong. Volumetric extrusion is the better way to do extrusion moves, and most likely that's what all printers will eventually end up doing. It ought to work for what will likely be the standard.

2) You're right, a lot of slicers won't have a setting than can be read from the gcode. I think the right thing to do is to just have a checkbox in the printer profile in OctoPrint that says `interpret E movements as volume` or `use volumetric E` or whatever. There's already a printer profile with nozzle diameter, bed size, and so on. It would be trivial to just add a checkbox.

3) I actually don't think the filament diameter is an issue. Even though you might do something like `M200 D1.73` to compensate for thinner filament, the volume calculation will still be correct if you're calculating it with a value of 1.75 (in fact the only reason you'd want to set a smaller filament diameter is to try and keep the volume the same). Technically your length calculation would be slightly off (and only very slightly), but at that point measuring by length doesn't mean much. If you have a 1kg spool of 1.75mm PLA you're going to calculate that you have about 333m of filament. So if OctoPrint says you've printed 10m of filament, you're going to think "I just used 3% of my spool". If you measure the diameter at 1.73mm, that means your 1kg spool is actually 340m and you've actually printed 10.2m of filament. You've still used 3% of your spool, and unless you've unwound and physically measured the length of your filament, there's not much point in trying to figure out the exact measurement of length. The filament is out of spec anyway.

I think the best way to do it would be to put a `use volumetric extrusion` checkbox as well as a filament diameter field in the printer profile. That way there's no need to rely on the slicer recording it's settings in the gcode, and if you don't turn the setting on nothing changes for non-volumetric printing. And you could even adjust for the small differences in filament diameter by changing the diameter in the printer profile. Although, again, it probably doesn't matter.

mikenew12, 2016-08-19:
I'd also be happy to do a PR if @foosel wanted to weigh in on how it should work.

foosel, 2016-08-24:
Haven't weighed in yet because frankly I haven't had the time yet to think it through properly. But I guess the printer profile parameter would actually be my own approach to that whole problem. Filament diameter field is a bit tricky though - you don't actually want to to adjust the printer profile every time you have a slight variation. Then again, you wouldn't actually need to do that to still get somewhat reliable values, it should be made absolutely clear though that this is a "rough" value, so something like a drop down rather than a float value input field ("2.85mm", "1.75mm" - possibly per tool?). 

But: I'm currently in feature freeze mode for 1.3.0 because otherwise I'll never see the end of that, so if you do send a PR, I won't be able to merge that in until after that's released (I might migrate 1.3.0 prep work to another branch than `devel` though while it's in RC phase and open up `devel` as new 1.4.x to allow earlier merges).

mikenew12, 2016-08-26:
@foosel that makes sense to me, although I'd like to give the user the option to set the filament diameter to a more precise value if they want to. So if you got a roll that ran a little thin, you could set it to 1.72mm and get a better measurement. Although I also don't think the length means much. Filament is sold by mass, not length, and I'd imagine that anyone tracking length of filament used is really just interested in what % of the spool they've used up or in the volume of plastic they've printed, and for that there's no need to tweak the filament diameter.

So maybe a dropdown with 1.75mm and 2.85mm, but also allow direct input? I hadn't thought about doing it per-tool but I guess that would make sense.

No worries about the feature freeze. I don't think there's any big hurry to get it merged.

foosel, 2017-03-08:
Turned into a request to support volumetric extrusion.

@michaelnew did you send a PR and I managed to completely overlook it the past months or did you never get around to that after all? No worries if not, I just want to make sure nothing got buried.

michaelnew, 2017-03-09:
@foosel I haven't :( 

I still want this and I still plan to. I have some time coming up and I'm hoping to get to it.