OctoPrint/OctoPrint #1433: Newly loaded model can't be printed until restarting OctoPrint

sfsdfd, 2016-08-01:
#### What were you doing?

I designed a model using GCode, uploaded it in OctoPrint, and waited for OctoPrint to finish analyzing it. I then pressed the Print button.

#### What did you expect to happen?

I expected OctoPrint to indicate that the print had started, and my printer to start behaving accordingly.

#### What happened instead?

OctoPrint changed the status of the printer to "Printing," and the textbox in the Terminal tab indicated that the status of my printer had changed from "Monitoring" to "Printing." However, nothing further happened. The printer didn't even set the target temperatures of the extruders and bed - it just stayed in the "Printing" phase without any appearance of activity, indefinitely.

After I restarted OctoPrint and clicked "Load and Print" for the same model, OctoPrint successfully sent the model to the printer, and it started printing as expected.

So it seems that there may be some kind of bug with the process of uploading and analyzing models: after completing this process, OctoPrint is not quite ready to print it.

#### Branch & Commit or Version of OctoPrint

1.2.15 (master branch)

#### Printer model & used firmware incl. version

FlashForge Creator Pro, firmware version June 2015

#### Browser and Version of Browser, Operating System running Browser

OS X (El Capitan) 10.11.5; Google Chrome 52.0.2743.82 (64-bit)

#### Link to octoprint.log

http://pastebin.com/Nda952Tk

#### Link to contents of terminal tab or serial.log

(serial.log is null; terminal tab contents during incident not available)

#### Link to contents of Javascript console in the browser

(Javascript console contents during incident not available)

#### Screenshot(s) showing the problem:

(not available)

I have read the FAQ.

markwal, 2016-08-01:
This could be an issue with the GPX plugin.  If you're able to reproduce, that terminal tab contents would be really helpful at tracking it down.

foosel, 2016-08-28:
Any news on those terminal tab contents?

sfsdfd, 2016-08-31:
Thanks for following up. I've encountered a few more instances of the "Printing But Not Really" problem... unfortunately, I haven't captured relevant terminal tab info yet.

Part of the problem is that, honestly, I don't keep as close an eye on my printer as I probably should. When it occurs, usually the printer has been sitting there in this zombie state for quite some time - and I don't actually know how far back the terminal tab queue goes.

The good news for debugging purposes is that I'm printing *a lot* these days, and this incident is happening often enough that I should be able to reproduce it and capture the terminal tab output. I'll report back with results.

One item that we can rule out: I suspected a problem with the "Automatically Load Print" plug-in, where it was not quite succeeding in handing off a model to the printer - such that clicking Print made OctoPrint think that the printer was ready to go, whereas the printer had only part of a print that it couldn't use, etc. However, this bug has continued to occur after I uninstalled that plug-in.

sfsdfd, 2016-09-10:
Okay - I've got some results to report. It's not the same circumstances as above, but the same problem that I described above occurred. I captured the terminal tab contents and have them for your review (I don't know how much help it will be, though...)

I had asked OctoPrint to start a print - in case it matters for some reason, here's the model that I was using:

http://www.djstein.com/Misc/model.gcode

A moment after initiating the print (still into the warmup stage), I canceled it via OctoPrint because I wasn't sure I had used the correct bed temperature. I confirmed the temperature, and then asked it to print again. As the terminal tab contents indicate, the status changed from "Monitoring" to "Printing," but nothing further happened. Here's the terminal tab contents:

http://pastebin.com/2QP69gaz

I then restarted OctoPrint, reloaded the model, and clicked Print. This time everything worked fine:

http://pastebin.com/Tfvm7z3m

The only difference was having previously canceled the earlier print. Seems that this causes OctoPrint to enter this zombie state, where it thinks it's printing but the printer is sitting there idle.

I've also noticed a more significant version of this issue, where OctoPrint not only thinks that the printer is printing, but it reports layer-by-layer results. The layers are displayed at an unbelievable pace - something like 2-3 seconds per layer - while the printer is sitting there doing nothing. I'll capture and present the terminal tab contents when it next occurs.

sfsdfd, 2016-09-11:
Okay - I have results. It's a slightly different scenario than I described above, but similar behavior.

Check this out:

http://pastebin.com/QQLGAWck

I loaded a print and then canceled it because I wanted to print it a little differently. I uploaded the new print, and then clicked Print - and... OctoPrint just sat there. It changed the status from "Operational" to "Printing," and did nothing else.

I restarted OctoPrint and then loaded and printed the same model, and got this instead:

http://pastebin.com/pc2Tbh1R

...which is running as expected.

In general, OctoPrint seems to have some non-deterministic behavior when you cancel a print, either via OctoPrint or via the printer:

* Sometimes it works OK. 

* Sometimes (frequently), it does this zombie-state thing the next time you ask OctoPrint to print anything.

* Sometimes, when OctoPrint starts the next print, it actually reports that the print is progressing. The status changes to Printing, and layers fly by very quickly. But when you look at the actual printer... nothing is happening; it's sitting there in an idle state. Seriously, this has happened maybe five times. Very weird. I'll definitely send terminal tab contents the next time I catch it in the act.

* Once - yesterday - I canceled the print via the printer LCD, and my bed and nozzle returned to their idle positions... yet, the nozzle kept twitching *and extruding in mid-air*. Apparently, OctoPrint hadn't gotten the message, and was still streaming instructions to it after the print was canceled.

foosel, 2016-09-12:
Taking a look at that first log:

```
Send: N14 M140 S110 T0*20
Recv: ok
Send: N15 M134 T0*85
Recv:  T0:28 /0 T1:27 /0 B:19 /0
Recv:  T0:28 /0 T1:27 /0 B:18 /110
Recv:  T0:28 /0 T1:27 /0 B:20 /110
Recv:  T0:27 /0 T1:27 /0 B:23 /110
Recv:  T0:28 /0 T1:27 /0 B:23 /110
Recv:  T0:27 /0 T1:27 /0 B:25 /110
Changing monitoring state from 'Printing' to 'Operational'
Recv:  T0:28 /0 T1:27 /0 B:27 /0
Recv:  T0:27 /0 T1:27 /0 B:27 /0
Recv:  T0:27 /0 T1:27 /0 B:25 /0
Recv:  T0:27 /0 T1:27 /0 B:25 /0
Recv:  T0:28 /0 T1:27 /0 B:25 /0
Recv:  T0:27 /0 T1:27 /0 B:23 /0
Recv:  T0:28 /0 T1:27 /0 B:24 /0
Recv:  T0:28 /0 T1:27 /0 B:22 /0
Recv:  T0:28 /0 T1:27 /0 B:22 /0
Recv:  T0:27 /0 T1:28 /0 B:23 /0
Recv:  T0:27 /0 T1:28 /0 B:23 /0
[...]
```

OctoPrint tells the printer to heat up. It's a blocking heat, so the printer will not return from this operation until it has reached the target temperature, but will log current temperature output every couple of seconds while it's heating up. Taking a look at those logs though it's obvious that the printer has neither a target temperature set, nor is it actually heating up as a cause of that. So that `M140` will block ad infinitum.

There's nothing OctoPrint can do here. It relies on the printer to do what it's told, especially with blocking commands like `M140`. Since this is a FlashForge printer which needs the GPX plugin to function at all, maybe @markwal should take a look at how the plugin is handling cancelled heatup requests. Might be that something is going wonky in the plugin there, tracking the state and doing the translation.

> In general, OctoPrint seems to have some non-deterministic behavior when you cancel a print, 
> either via OctoPrint or via the printer:

>  * Sometimes it works OK.

Yay!

>  * Sometimes (frequently), it does this zombie-state thing the next time you ask OctoPrint to print anything.

Caused by the firmware/the GPX plugin entering a never ending heatup-but-not-actually-heating-up cycle, nothing I can do about that, needs to be fixed either in the firmware or the GPX plugin.

> Sometimes, when OctoPrint starts the next print, it actually reports that the print is progressing. The status changes to Printing, and layers fly by very quickly. But when you look at the actual [...]

That also sounds like something either in the firmware or the plugin going wrong. Unless something in the firmware tells OctoPrint something went wrong ("Error"), it will assume it can proceed streaming the file to it (which is what "printing from OctoPrint" boils down to, more on that below). My guess would be that for whatever reason the commands from OctoPrint are accepted by plugin/firmware, but nothing is done with them, which leads to the very first "ok done" turnaround you are seeing.

Nothing I can do about - OctoPrint expects the printer on the other end of its serial line to actually do what it's told before it sends an `ok` back, signaling to OctoPrint that it can send the next line. If it doesn't do anything but doesn't trigger an error message or something like that telling OctoPrint about the fact that it's refusing to work (and ideally also why), there's no way I could detect that, let alone solve it.

> Once - yesterday - I canceled the print **via the printer LCD**, and my bed and nozzle returned to their idle positions... yet, the nozzle kept twitching and extruding in mid-air. Apparently, [...]

Unless the printer reports back when something on its LCD is pressed AND @markwal had his plugin translate that into an [action command](http://docs.octoprint.org/en/master/features/action_commands.html), it's completely unsurprising that OctoPrint didn't know you'd hit cancel somewhere outside of it.

When OctoPrint is printing a file, all it does is streaming a lot of GCODE very fast to the printer. In GCODE based printers there's no way to tell the printer "listen, I'm as the host am now going to start printing a file, so anything you receive from me from now on is part of that file, not just manually entered commands". Hence, the printer usually doesn't know if what it has on the serial line is printing or just sending a lot of commands very fast. No difference in the protocol there.

I don't know how Sailfish based printers handle that, if they have a "printing" state that can somehow be triggered via some kind of serial command or not, and I also don't know what actually does happen when you click "Cancel" on their LCD while arbitrary commands are being sent to serial - my expectation would range from "nothing" to "the printer simply does some cancel sequence but doesn't notify the host about that".

With that information in place, it should be obvious why OctoPrint "hadn't gotten the message" - the printer never sent it, since there is no back channel (not by my choice btw, I also run into similar problems all the times with GCODE based firmwares where people design fancy long running menu operations and don't even think about the consequences of those for anything connected via serial). 

As a general rule of thumb, either control your printer through OctoPrint OR your LCD. If you mix the two you might produce effects as the above, because you basically have your printer do something behind OctoPrint's back - it's obvious that that can mess with things.

Again, just to make this perfectly clear, OctoPrint does support [action commands](http://docs.octoprint.org/en/master/features/action_commands.html) from the printer, and if it sends them on LCD pause, cancel, whatnot commands, it will do exactly what it should be doing. But it needs the firmware to cooperate for that to work and most firmwares (neither in the GCODE world nor as far as I know with Sailfish) don't.

All in all, all of those issues sound like they are caused by either the firmware or the GPX plugin or both, OR are expecting something to work which sadly can't work unless the *firmware* does its part (OctoPrint already does its part).

sfsdfd, 2016-09-12:
Wow! Thank you for the extremely detailed explanation.

I completely understand the cause-and-effect exchange that you laid out - but I have some ideas.

Let's start with this:

> Caused by the firmware/the GPX plugin entering a never ending heatup-but-not-actually-heating-up cycle, nothing I can do about that, needs to be fixed either in the firmware or the GPX plugin.

Well... sort of. I think that there's a gap between the expectation, and the actual behavior of the printer.

As you've described it, the protocol permits a blocking call to set the temperature. In accordance with that protocol, OctoPrint expects the printer to comply; that the printer will do nothing else until the instruction has been achieved; and that the printer is completely inaccessible for additional instructions until it reports success.

But that isn't actually how the printer behaves, for two reasons:

1) As I noted in the second PasteBin above - after the printer had entered this zombie state and I reset OctoPrint, the printer was completely ready to receive additional instructions. The deadlock was broken by restarting OctoPrint and reconnecting. So the blocking wasn't really the printer; it was OctoPrint's instruction feeder. 

2) You wrote that "there is no backchannel" for OctoPrint to indicate a change of plans - but there actually is, right here:

Recv:  T0:27 /0 T1:27 /0 B:25 /110
Changing monitoring state from 'Printing' to 'Operational'
Recv:  T0:28 /0 T1:27 /0 B:27 /0

OctoPrint has received both the message about the status change - and (more subtly) an indication that the target temperature was set to 110, but subsequently reset to 0. Either (or both) of those events could be hooked as an indication that something has changed, and OctoPrint could update its status accordingly.

Based on those observations, I think that there's an easy opportunity to make OctoPrint more robust by parsing and responding to these messages. The closed-feedback loop shouldn't solely await a "target temperature achieved" reply; it could monitor the actual progress for significant status changes. That might be applicable not only during warmup, but throughout the printing cycle: comparing the expected responses of the printer with actual responses to detect unexpected status changes.

Now, I recognize that this approach may be a non-starter for many reasons. This might be a printer-specific behavior, and even a breaking departure from the formal protocol. It might not occur consistently (such as the circumstances of the other anomalous results that I described). It might be too complicated to warrant attention, or simply a lower priority than many other issues. But it might also be an easy fix that expands the stability of OctoPrint for a broad range of printers. (I definitely don't know enough about OctoPrint, GPX, etc. to make any guesses - just outlining the possibilities.)

Hope this is helpful. Thanks for your attention.

(A final comment: I would be thrilled to contribute some code to OctoPrint. It's a project of a manageable scope, and I would just need a free weekend to dig into the code and attain some familiarity. Unfortunately, it doesn't look like that weekend is coming any time soon... hopefully some day! Until then, I'll have to limit my contribution to Patreon and end-user feedback. Will try to make the latter as high-yield as possible.)

Regards,

David Stein

markwal, 2016-09-12:
This is entirely a bug in the GPX plugin.  It has to be leftover wait state from the cancel.  I haven't figured it out yet or how to get a repro, but there's enough info here to pin in clearly on my plugin.  Moving it to the GPX plugin issue tracker.

closed 2016-09-12 by markwal

foosel, 2016-09-16:
@sfsdfd I was out of office for a couple of days due to personal reasons, but I still wanted to not leave your long write up uncommented - sorry for the delay!

> But that isn't actually how the printer behaves, for two reasons:

> 1) As I noted in the second PasteBin above - after the printer had entered this zombie state and I reset OctoPrint, the printer was completely ready to receive additional instructions. The deadlock [...]

> 2) You wrote that "there is no backchannel" for OctoPrint to indicate a change of plans - but there actually is, right here:
> ```
> Recv: T0:27 /0 T1:27 /0 B:25 /110
> Changing monitoring state from [...]

There's a misunderstanding here :) The "Changing monitoring state" line comes from OctoPrint. OctoPrint has an internal state model, which is what is reflected in the state panel. One of those states is "operational" (as in, connected to the printer but idle), another is "printing" (as in, either pushing lines from a local file to the printer, or having instructed the printer to print something from its own SD card). The printer doesn't know the difference between the two, there's no way to tell it "hey, you are now printing!" (or at least nothing even remotely standardized that I'm aware of). Stuff with `Recv` in front is coming from the printer, stuff with `Send` in front is sent to the printer, everything else is stuff injected into the terminal log by OctoPrint to allow for some context.

So, what actually went over the line is only this:

```
Send: N14 M140 S110 T0*20
Recv: ok
Send: N15 M134 T0*85
Recv:  T0:28 /0 T1:27 /0 B:19 /0
Recv:  T0:28 /0 T1:27 /0 B:18 /110
Recv:  T0:28 /0 T1:27 /0 B:20 /110
Recv:  T0:27 /0 T1:27 /0 B:23 /110
Recv:  T0:28 /0 T1:27 /0 B:23 /110
Recv:  T0:27 /0 T1:27 /0 B:25 /110
Recv:  T0:28 /0 T1:27 /0 B:27 /0
Recv:  T0:27 /0 T1:27 /0 B:27 /0
Recv:  T0:27 /0 T1:27 /0 B:25 /0
Recv:  T0:27 /0 T1:27 /0 B:25 /0
Recv:  T0:28 /0 T1:27 /0 B:25 /0
Recv:  T0:27 /0 T1:27 /0 B:23 /0
Recv:  T0:28 /0 T1:27 /0 B:24 /0
Recv:  T0:28 /0 T1:27 /0 B:22 /0
```

My guess is you clicked cancel when the bed target jumped from 110 to 0/the state change was logged, but the printer (or rather the GPX plugin) didn't get out of its heatup-or-not-loop. That is also what can happen with GCODE based printers running Marlin and the like which get set to a heat-and-wait of a temperature they cannot reach, e.g. 150degC on the bed with a fan blowing over it. In such cases, the printer will simply become completely unresponsive from the host's point of view and there's nothing the latter can do about that besides maybe disconnect and hope that the controller is wired up to reset on serial connect (not all of them are, which means someone has to physically reset the printer in such cases).

> Either (or both) of those events could be hooked as an indication that something has changed, and OctoPrint could update its status accordingly.

Only the target temperature change is something that actually could be hooked here, but yes, I agree. The problem though is that IF the printer actually gets stuck in such a loop, be there a temperature change or no change, at least in all GCODE based firmwares there would be no getting out of it anymore without resetting the controller. So the only thing that would allow us is "if it gets stuck in the heatup loop too long without significant change, disconnect and signal an error" [which opens up the tricky question "what is significant?"]. All bidirectional communication with the printer ceases once it enters such a heatup loop, there's no recoverable way to signal a stop. I think stuff like this is being discussed in current Marlin, but that leaves out the large majority of unupdated printers running unmaintained forks out there.

So, the only feasible thing with the current situation "on the market" would be to simply give up and disconnect, ideally also force an `M112` emergency stop down the serial connection in the hopes that still fits in the serial buffer *and* be processed (if that works depends entirely on the exact firmware implementation though). And that would depend on the target temperature to even be monitorable - some firmwares only output the current temperature during heatup, not what their target is. Which would leave us with "fully monitor the temperature fluctuations of the heaters compared to their targets at all times" and that would start to become a bit too much of trying to be intelligent and doing the firmware's job. Still, might be something that could be done via a plugin. The more intelligence like that in the comm layer, the more even a slight deviation in behaviour by FancyNewFirmware can break stuff in unexpected ways. 

I'd rather leave the firmware tasks to the firmwares (and try to get their developers to widen their point of views to also think about how their interfaces are used).

Long story short: I agree with your analysis that stuff *could* be made more solid here for a handful of very specific error cases, but the majority of endless-heatup-loop manifestions would not be solvable by that. Staying responsive during a heatup loop *inside the firmware* would solve this, and even better, detecting on the firmware side when a target temperature cannot be reached, but as long as that is not the case, all the host can do is hope stuff doesn't get stuck and leave it up to the user to disconnect (because disconnecting when the heater just is horrible slow is also not really an option that people will be happy with I fear).