OctoPrint/OctoPrint #1434: Special Characters in Filename return 404

evils, 2016-08-02:
I couldn't find this elsewhere and I believe this was working before, so here I am reporting it!

#### What were you doing?

Created a .gcode file with "[]“ in the filename through S3D and uploaded it via scp to my octoprint.
Tried loading/printing the file resulted in no error message, yet the console gave me this:

`packed_libs.js?35d69079:5  http://primary-pi2.local:5000/api/files/local/%5B50Fan%5DTestcube20x20x10_Hole10-ABS.gcode 404 (NOT FOUND)send @ packed_libs.js?35d69079:5ajax @ packed_libs.js?35d69079:5GcodeFilesViewModel.self.removeFile @ packed_app.js?f9e76fc5:1134(anonymous function) @ VM536:3(anonymous function) @ packed_libs.js?35d69079:1020dispatch @ packed_libs.js?35d69079:4y.handle @ packed_libs.js?35d69079:4`

This wouldn't be as much of a problem if it wouldn't mess with the delete function.

#### What did you expect to happen?

File being able to print

#### What happened instead?

UI: No Error
Console: 404 Not Found

#### Branch & Commit or Version of OctoPrint

1.2.15 Master Branch

[Can be found in the lower left corner of the web interface.]

#### Printer model & used firmware incl. version

Marlin 1.1.0 - RC7 + RCBugFix

#### Browser and Version of Browser, Operating System running Browser

Latest Chrome on latest Mac OS X (non beta)

#### Link to octoprint.log

http://pastebin.com/jcT2zLHw

#### Link to contents of terminal tab or serial.log

For whatever reason, serial.log is empty

#### Link to contents of Javascript console in the browser

http://pastebin.com/gRX9Vhgu

I have read the FAQ.

foosel, 2016-08-15:
> and uploaded it via scp to my octoprint

Please don't do this. When uploading through OctoPrint, OctoPrint will make sure to strip problematic characters from filenames and do some housekeeping it needs to be able to properly operate. You are basically circumventing this, interfered with internal operations and this is what broke stuff here (not only file deletion, but also the internal tracking of metadata). Use the watched folder for uploading via SCP or other means if you don't want to go through the upload API, that's what it's there for. Do not perform manual changes in the uploads folder.

That being said, I just pushed a commit to the `maintenance` branch (see above) that will take care of sanitizing (and housekeeping) any files in the uploads folder that are not yet sanitized, that should auto-correct such problems in the future on each file list (by renaming and indexing the offending files).

evils, 2016-08-15:
@foosel Thanks for taking time to respond. I had a feeling that the scp upload was breaking my setup causing this issue but I wasn't super sure.

Normally I would close this issue now, but as you were kind enough to take care of even the last person trying to break your great system (me) in the maintenance branch, I leave this open until maintenance gets into some release and this issue will be fixed then.
Otherwise, feel free to close!

Thanks again

foosel, 2016-08-28:
Commit mentioned above has since been merged to devel as well, closing.

closed 2016-08-28 by foosel