OctoPrint/OctoPrint #1438: Babystepping Z-Axis?

jstefanop, 2016-08-05:
I know some control software like Simplify3D and MatterControl can BabyStep while printing...is this possible with Octoprint? Currently the control buttons gray out after printing starts, but maybe a simple plugin can be written that enables babystepping control during printing?

(best use for this is during first layer, and fine tuning the first layer while its printing)

gddeen, 2016-08-05:
It is kinda tough because the printing is done in absolute Z references rather than relative. So, to skip up, the code would need to know the z level... which is real tough because there is a comand queue on the printer... this buffers commands so noone knows where z level is.

There are commands to change the level of the platform, but they are dependent on auto bed leveling or mesh bed leveling.

There are programs to go through the gcode and switch it to relative coordinates. Then it would be easier.

Im ALL for it. Anything, any  partial solution.

An easier solution might be to put an up down button on printer and write the changes into firmware. Give command in printer to push those buttons have octop give command...

Really good luck...

closed 2016-08-20 by GitIssueBot

tweichselbaumer, 2017-10-29:
There is now a Gcode (M290) added to the Marlin firmware. https://github.com/MarlinFirmware/Marlin/pull/8014

So you can add a custom control to the config.yaml or maybe someone will write a plugin to control the babystepping during printing.

jlipavsky, 2018-01-31:
ok the question is..... on the new prusa mk3, right on the lcd is live z height correction during a print.... from what i hear the same thing is possible in other gcode sending software, the question is can this be implemented into octoprint so z calibration can be done on the fly while printing in underway.... the model of printer is irrelevant

tweichselbaumer, 2018-02-01:
I don't know if the prusa firmware provides the command M290 right now. But if you use the original MarlinFirmware >= 1.1.7 it should be possible the change the height during printing from octoprint. 

From my point i can say, it works with the code i pushed with the referenced request. The development team did some change but i think they only extend the feature. For my printer i added some more buttons to Octoprint via the config.yaml as descriped here: https://github.com/foosel/OctoPrint/wiki/Cookbook:-Custom-Controls
Now i can move the z-axis while printing.

JTT-AE, 2019-04-22:
> I don't know if the prusa firmware provides the command M290 right now. But if you use the original MarlinFirmware >= 1.1.7 it should be possible the change the height during printing from [...]

Share your changes!

tweichselbaumer, 2019-04-26:
> Share your changes!

Which one? I did a pull request to the MarlinFW so everything is shared. 

http://marlinfw.org/docs/gcode/M290.html
http://docs.octoprint.org/en/master/features/custom_controls.html

```
 - name: Babystepping
    layout: horizontal
    children:
          - name: Z +1mm
                command: M290 Z1.0
           - name: Z -1mm
                command: M290 Z-1.0
           - name: Z +0.1mm
                command: M290 Z0.1
           - name: Z -0.1mm
                command: M290 Z-0.1
```

Shine6Z, 2020-04-20:
> I don't know if the prusa firmware provides the command M290 right now. But if you use the original MarlinFirmware >= 1.1.7 it should be possible the change the height during printing from [...]

In fact, if the M290 is to take effect during printing, it must have an interface that supports the queue-jumping command. Right now, I just can't do queue jumping, so I might damage the printer.