OctoPrint/OctoPrint #1439: Signals Putting BLTouch in Error in Middle of Print

timeshell, 2016-08-05:
#### What were you doing?

Printing from Raspberry Pi with customized Linux OS and Octoprint installed on top.  Printing to a Cobblebot Basic using Arduino/RAMPS14.  Slic3r 1.2.9 generated the gcode file on the PC which was uploaded to the Octoprint server.  Probing using a BLTouch.

#### What did you expect to happen?

Expect the print to finish without the BLTouch going into error condition.

#### What happened instead?

The BLTouch goes into an error condition within an hour of the print starting.  The movement of the printer then drops the pin causing it to collide with the printed object and stuff starts going wrong.

Why am I bringing this up here?  Well, it doesn't seem to happen when printing from the PC instead of the RaspberryPi using the exact same gcode through Repetier-Host.  So, it seems to be something coming through from the Raspberry Pi/Octoprint server.

#### Branch & Commit or Version of OctoPrint

Latest.  Has been occuring through the last 3 versions.

#### Printer model & used firmware incl. version

Cobblebot Basic / Marlin 1.1.0-RC4

#### Browser and Version of Browser, Operating System running Browser

N/A

#### Link to octoprint.log

http://pastebin.ca/3674336

#### Link to contents of terminal tab or serial.log

Empty

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

N/A

I have read the FAQ.

Salandora, 2016-08-05:
Uhm well I don't really think thatoctoprint is the cause of the problem.

Can you please explain your setup a bit more especially the hardware part. What psu do you have,  what pi, how is the sensor connected to your ramps board e.g. maybe some little drawings.

Actually I do not have such a probe nor do I know how it gets connected or what it is capable of.
So I can just guess a bit around and make some suggestions depending on the informations I receive.

jneilliii, 2016-08-05:
I've seen a similar issue with my BLTouch.  Not sure if it is exactly the same, as my probe stays retracted but the led on it starts flashing after the print starts as if in alarm mode.  It doesn't take much time for it to happen for me, and seems to be within the first couple of layers, not after a long period of time.  The only thing I can think of that has changed with my gcode is adding "; Slice" comments on layer change in order to post-process the file with a "wood grain" script to change the extruder temperature to get that effect. I"m able to use my custom control buttons to reset the probe while it's printing, so not a huge issue for me. Attached is a gcode that this has happened to me with.

[beziervase_3_wood_grain.gcode.txt](https://github.com/foosel/OctoPrint/files/404689/beziervase_3_wood_grain.gcode.txt)

timeshell, 2016-08-05:
Did it happen every time you did that print?  Mine happens 9 times out of ten.  But on the PC it works just fine.

I'm using a standard RAMPS setup.  The BLTouch is connected to the Z end stop and first set of servo pins.  I thought it might be the PSU and changed to another one, an ATX, to see if that would correct it.  The first was a non-ATX.  The Raspberry Pi connects to the Arduono via USB.

jneilliii, 2016-08-06:
So I just did a brand new slice of another object without adding any special post-processing and it did it again.  I didn't catch exactly when it did it, but for me it was at the beginning of the print during heat-up, either bed or extruder. I'm going to take out the "; Slice" lines from my layer change custom gcode and see if it still does it.

timeshell, 2016-08-06:
Did you try it from PC instead of the Raspberry Pi?  If so, did it work properly?

Salandora, 2016-08-06:
Well I actually have 2 ideas:

Could you both make a OctoPrint installation on your PC and do a print out of it. 
Just to check if the problem comes through OctoPrint

The second one is more or less a very wild hope: 
I'm assuming that your boards get powered through an ATX or other sort of PSU, so the Ramps board and the Rambo should create their own 5V supply. 
But the PI's USB port also gives an extra 5V rail through the Raspberry Pi, and my guess here is that the 5V of the Pi is having some noise in it which also gets to the Servo pin and then sets the Sensor into an error mode.
If any of you have an Oscilloscope could you try to check the 5V rail while printing?
And maybe the signal pin also?
If none of you have one, do you by any chance have an USB cable without the 5V rail, or some kind of adapter.

timeshell, 2016-08-06:
I actually already put a scope in the signal pin during the print.  It was a pretty clean signal.  I was wondering about signal quality from the PI though.  I will try the PC local Octoprint for giggles.  That's a good idea.  However, I might add that both the PI and the servos in my installation are being fed from 5V sources.  I will also try powering them both from the same ATX PSU.

timeshell, 2016-08-06:
Oh, and in case anyone wonders, we can't disable the servo during print.  It will cause the pin to fall for sure.

jneilliii, 2016-08-06:
So my last test removing "; Slice" on layer change in my slic3r there was no issues during the print, however it was a print I did as I went to bed and unfortunately upon waking up I have a nozzle extremely jammed up with wood filament.  Not sure if I'll be able to do additional testing for a while since I don't have any extra nozzles laying around.

Salandora, 2016-08-06:
I have googled a bit around and well I like the device it's really nice.
I also found an image where it shows when the alarm mode get's triggered.
But I don't really understand that image xD (its on the indiegogo campaign site).
So I dont really have a clue why it would go into an error state or why it pushes the pin down mid print .

timeshell, 2016-08-06:
When it goes into error state, it appears to stop trying to hold the pin.  So the movement of the printer will then cause it to drop.  It is being held in place by a magnetic field.  But when it goes into error, it appears to ignore the servo state.  I am the official Canadian reseller btw.

Salandora, 2016-08-06:
hmm "According to the manufacturer, it also goes into error state if it senses that it can no longer hold the pin", could it be possible that the servo 5V rail is droping? 
Could you attach a Scope and an Amperemeter to the 5V rail of the servo connector and check if it stays constant?

timeshell, 2016-08-06:
While that may be the design, the BLTouch can be reset during the print and continues to do so after being reset.  I have also seen the case where it goes into error in the middle of the print and doesn't drop the pin, at least not right away.  So it appears something else may still be triggering the error.

I just completed a print using Octoprint on the PC which did complete successfully.  My next print will be with a new RaspPi 3 on the same ATX PS as the Arduino / RAMPS.

timeshell, 2016-08-10:
Ok, I just did a print using Octoprint via my PC and within the first few layers the BLTouch went into error.

Salandora, 2016-08-11:
Uhm that's bad.
I have no clue what could cause this problem.
Hmm maybe it'll help if you provide a serial log.
So we can see what goes through the line when the device goes into error mode.
But I'm still sure that Octoprint is not causing the problem. 
Because how should it? 
Octoprint is just sending over the line what is in your gcode. And some Temperature pollings.

timeshell, 2016-08-11:
I am fully in agreement.  But so far, in all my tests after coming across this issue, that appears took be a common connection.

I'll work on getting a serial log.

timeshell, 2016-08-17:
This isn't an Octoprint issue.

closed 2016-08-17 by timeshell

foosel, 2016-08-17:
Just out of curiosity - what is the cause?

timeshell, 2016-08-17:
I don't know yet.  But I have determined that it occurs on the PC in Repetier also now.  The next thing I'm thinking of trying is running 2 BLTouches in parallel, one horizontally, and see if they both go into error at the same time, or at all.

Salandora, 2016-08-18:
Good idea. Maybe it's a faulty device.
Hope you will find out what it is.

jneilliii, 2016-08-18:
What's the chance of it being caused by some other firmware related issue.  Something like thermal run away protection or something similar?  I have been printing now without any issues and mine has not alerted since those first couple of times it did it initially and I haven't changed anything outside of working on my plugins.

timeshell, 2016-08-18:
I have had it happen with Marlin 1.1.0-RC4 and RC7 on both PC and RPi.  I wonder if it is a glitch in the Arduino too.  I've read about servo jittering and since the BLTouch uses the servo output I'm wondering if it's getting signals that are confusing it.  I'm about to try a new printer configuration with a different BLTouch.  We'll see where it goes.

Grogyan, 2016-08-19:
@timeshell  The BLtouch uses a static magnetic field to hold the pin in
place.  If the pin is dropping mid print, then there is a problem with your
particular BLTouch sensor.
I have one of the early versions, and a Rev2 both have weak fields holding
the pin in place, and to me it is sufficient.
Marlin 1.1 R7 has added support for BLTouch, but this won't fix weak static
magnetic fields

timeshell, 2016-08-19:
This isn't entirely the issue.  When the M280 p0 s90 is issued, the servo state should hold the Lin up.  Even if you pull the pin down it is supposed to pull it back up.  If it can pull it back up, there is no reason why it should be able to drop in the first place.

Grogyan, 2016-08-19:
I think you misunderstand, once the command is issued, the magnetic coil is
no longer energized, and the pin is held in place by the static magnetic
field present.

In anycase, such discussions about BLTouch should be in the Marlin GitHub,
nothing to do with OctoPrint.  As the commands can be issued by any host
program.

Please also contact the BLTouch guys.

timeshell, 2016-08-19:
I'm only responding because I getting responses.... :p

Also, if the coil is no longer energized, it would be ok to disable servos once the probing is complete.  From the instructions from ANTCLABS, it isn't ok to disable the servos with the BLTouch.

I am in direct contact with ANTCLABS on this issue already.  I won't be posting any further here.

foosel, 2016-08-19:
@timeshell personally I found this very interesting to follow. The ticket is closed, so it's clear that this is not something i have to become active on myself, but it might be helpful for people discovering this thread through Google. So please, once you figure out what is the reason here, post it. 

Whoever feels bothered by the noise can use the Unsubscribe button to the right :)

[image]

timeshell, 2016-09-10:
I don't have 100% confirmation yet (not enough testing), but it appears that the issue all along may have been a poor power wire connection (bad pin on extension cable to BLTouch), just poor enough to trigger an occasional error during movement of the print head.  I accidentally discovered it when performing other maintenance and when lifting the cable far enough I noticed the power going off on the BLTouch.

PEBKAC.  DOH.

foosel, 2016-09-10:
@timeshell Ouch :) Stuff like that hurts, I know the feeling all too well.