OctoPrint/OctoPrint #1440: Octopi and Raspberry Pi3 USB Stick

ecagatay, 2016-08-06:
Hi!

I'm using the Octopi in my 3D printer with a Raspberry Pi touch screen. Firstly, I want to thank you and your organization to develop the magnificent software!

I have some problems with the Octopi and I would ask you about them. The wireless access works fine, but I want to give ".gcode" file that is prepared in another slicer program to the Octopi's interface with USB memory stick (for example, my friend gives it to me and he want me to print an object). However, when I put the memory stick in USB A socket of the Raspberry Pi3, I can't see the ".gcode" file.

How can I solve this problem? Or, could you please update the Octopi for this case? This is important to access without computer or another device.

ecagatay, 2016-08-06:
#### What were you doing?

The wireless access works fine, but I want to give ".gcode" file that is prepared in another slicer program to the Octopi's interface with USB memory stick (for example, my friend gives it to me and he want me to print an object). However, when I put the memory stick in USB A socket of the Raspberry Pi3, I can't see the ".gcode" file.

#### What did you expect to happen?

I expect to access my USB memory stick and ".gcode" file in Octopi webrowser.

#### What happened instead?

I don't see any file from memory stick. There are just files from Raspberry Pi's microSD card.

#### Branch & Commit or Version of OctoPrint

I use version 1.2.13.

#### Printer model & used firmware incl. version

My custom printer with Raspberry Pi 3.

#### Browser and Version of Browser, Operating System running Browser

Firefox and Chrome.

I have read the FAQ.

evils, 2016-08-11:
This is not a potential bug but misconfiguration.
Octopi (which is also a different project and should be put in there) doesn't automount removeable media as most other RPI distros do.

You might consider educate yourself on how to set automount up.
The keywords you would search for would be "debian" "usb automount" "raspberry pi"

Salandora, 2016-08-11:
I have no idea if octopi would automount an usb device but even if it would you wouldn't see the files because octoprint doesn't know that you plugged an usb stick with gcode files in.

So basically you would need to mount any usb drive to the same folder, dunno if sub folders would work.
And configure Octoprint to watch this folder/these folders.

foosel, 2016-08-15:
Not a bug. A plugin request at most.

foosel, 2016-08-15:
And as a request it's a duplicate of #1192, therefore closing.

closed 2016-08-15 by foosel

linanw, 2017-02-03:
Sharing how did I successfully setup to auto show my gcode & stl file in an USB Memory Stick/USB Drive/Thumb Drive in OctoPrint file list, in case someone still needs detail steps. 
1. sudo apt-get install usbmount  -y
2. Create a sub-folder, ie. "USB", in your ".octoprint/uploads folder". If you use OctoPi, the folder is under "/home/pi"
3. Edit "/etc/usbmount/usbmount.conf" for MOUNTPOINTS="<here is your .octoprint/uploads folder path + a sub-folder name>", in my case is MOUNTPOINTS="/home/pi/.octoprint/uploads/USB"
4. Edit "/etc/usbmount/usbmount.conf" for FS_MOUNTOPTIONS="-o uid=<uid your OctoPrint is running with>,gid=<uid your OctoPrint is running with>", in my case is FS_MOUNTOPTIONS="-o uid=pi,gid=pi", which is default settings for OctoPi as well.
5. reboot
You are done. Your will see the similar as it's in the picture after reboot and plugged your USB Drive.  
[image]

rkozak, 2018-08-31:
I followed the procedure provided by linanw and I can see the files but I can’t upload the the external drive via the folder mounted on it. Is there a way to allow write acces to this folder?

juri74, 2020-05-16:
> Sharing how did I successfully setup to auto show my gcode & stl file in an USB Memory Stick/USB Drive/Thumb Drive in OctoPrint file list, in case someone still needs detail steps.
> 
>     1. sudo apt-get install usbmount  -y
> 
>     2. Create a sub-folder, ie. "USB", in your ".octoprint/uploads folder". If you use OctoPi, the folder is under "/home/pi"
> 
>     3. Edit "/etc/usbmount/usbmount.conf" for MOUNTPOINTS="<here is your .octoprint/uploads folder path + a sub-folder name>", in my case is MOUNTPOINTS="/home/pi/.octoprint/uploads/USB"
> 
>     4. Edit "/etc/usbmount/usbmount.conf" for FS_MOUNTOPTIONS="-o uid=,gid=", in my case is FS_MOUNTOPTIONS="-o uid=pi,gid=pi", which is default settings for OctoPi as well.
> 
>     5. reboot
>        You are done. Your will see the similar as it's in the picture after reboot and plugged your USB Drive.
> 
> 
> [image]

i followed your instructions. i have default options but i can't see my usb memory stick. even if i power on with the stick already installed
there are a lot of options in MOUNTPOINTS. should i delete all? if not where should i place the mountpoint, before or after? thanks

whoim2, 2020-08-06:
sudo systemctl edit systemd-udevd
insert:

> [Service]
> PrivateMounts=no

sudo systemctl daemon-reload
sudo service systemd-udevd --full-restart

need for new raspbian:
https://github.com/rbrito/usbmount/issues/2
https://unix.stackexchange.com/questions/330094/udev-rule-to-mount-disk-does-not-work/330156#330156

GerogeFu, 2020-12-24:
Still not good here.  Here is my situation : if i pulgin the USB stick before i boot raspi and i can see the files in USB folder. And then i plug out USB stick , and i re-entry(means back and enter) USB folder and the files still there and i click the file ,it shows "can't load files", but in octoprint web ui, they are not in the file list. Then i plug in USB stick again and i re-entry USB folder to click file , i still got "can't load files".