OctoPrint/OctoPrint #1445: Heated Bed does not go above 100°C since 1.2.15 Octoprint Update

WOAAP, 2016-08-11:
Dear all,
since the Octoprint Update to version 1.2.15 I cannot heat my bed from my Ultimaker Extended 2+ above 100°C. It does not matter if I set the settings in the gcode file or if I manually change the temperature via the Octoprint webinterface.

I also changed the temperature settings of the ABS printing temperature profile of the bed to 110°C.

Independently of the input method the Octoprint Webinterface seems to accept the temperature input, but ignores the command and caps the temperature at 100°C.

Does anyone know a solution to this problem? If not is it possible to downgrade the Octoprint version?

Thanks so much for feedback.

Salandora, 2016-08-11:
Please provide a serial log.
This is necessary here.

Also does your UM 2 has an lcd display?
What does it say there what is set as target temperature.

Salandora, 2016-08-11:
One more thing have you tried to roll back to 1.2.13?

WOAAP, 2016-08-11:
I would love to do the roll back, but I am unable to do this. Also simple googleing could not help me. So if you could briefly explain how this works, I would be very grateful.

Do you mean with serial log the standard octoprint log-files? The UM2 does have a LCD display but does not show the temperature - it just displays "printing via USB".

Salandora, 2016-08-11:
https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-roll-back-to-an-earlier-version-after-an-update

About serial log inside the settings is an option to enable the serial log. 
Or copy the terminal tab after you set the temperature.

Also a screenshot of your temperature tab could be useful.

WOAAP, 2016-08-11:
The Terminal Tab gives the following:

Send: M140 S110.000000
Recv: ok
Send: M105
Recv: ok T:29.4 /0.0 B:25.6 /100.0 @:0 B@:0
Send: M105
Recv: ok T:29.1 /0.0 B:26.0 /100.0 @:0 B@:127
Send: M105
Recv: ok T:29.0 /0.0 B:28.5 /100.0 @:0 B@:127
Send: M105
Recv: ok T:29.1 /0.0 B:30.0 /100.0 @:0 B@:127
Send: M105
Recv: ok T:29.3 /0.0 B:31.8 /100.0 @:0 B@:127
Send: M105
Recv: ok T:29.1 /0.0 B:33.4 /100.0 @:0 B@:127

The Temperature Tab does not show anything exotic. I fill the Bed field with 110, press set and it acknowledges the input, but does cap it to 100.

I activated the serial log, but do not know how to export the file. I am currently rerolling to 1.2.13 an see if this does anything.

WOAAP, 2016-08-11:
Ah I could extract the serial.log; the last few lines after changing the bed temperature to 110:

2016-08-11 16:48:06,096 - SERIAL - DEBUG - Send: M140 S110.000000
2016-08-11 16:48:06,112 - SERIAL - DEBUG - Recv: ok
2016-08-11 16:48:07,968 - SERIAL - DEBUG - Send: M105
2016-08-11 16:48:07,983 - SERIAL - DEBUG - Recv: ok T:25.9 /0.0 B:28.5 /100.0 @:0 B@:0
2016-08-11 16:48:12,972 - SERIAL - DEBUG - Send: M105
2016-08-11 16:48:12,982 - SERIAL - DEBUG - Recv: ok T:26.2 /0.0 B:28.2 /100.0 @:0 B@:127
2016-08-11 16:48:17,971 - SERIAL - DEBUG - Send: M105
2016-08-11 16:48:17,985 - SERIAL - DEBUG - Recv: ok T:26.2 /0.0 B:30.0 /100.0 @:0 B@:127
2016-08-11 16:48:22,971 - SERIAL - DEBUG - Send: M105
2016-08-11 16:48:22,986 - SERIAL - DEBUG - Recv: ok T:26.5 /0.0 B:32.1 /100.0 @:0 B@:127
2016-08-11 16:48:27,981 - SERIAL - DEBUG - Send: M105
2016-08-11 16:48:27,997 - SERIAL - DEBUG - Recv: ok T:26.5 /0.0 B:32.9 /100.0 @:0 B@:127
2016-08-11 16:48:32,985 - SERIAL - DEBUG - Send: M105
2016-08-11 16:48:33,007 - SERIAL - DEBUG - Recv: ok T:26.8 /0.0 B:34.6 /100.0 @:0 B@:127
2016-08-11 16:48:37,986 - SERIAL - DEBUG - Send: M105
2016-08-11 16:48:37,999 - SERIAL - DEBUG - Recv: ok T:26.3 /0.0 B:37.1 /100.0 @:0 B@:127

foosel, 2016-08-11:
According to this log, OctoPrint is telling your printer to heat up to 110°C, just like you told it to:

    M140 S110.000000

Your printer however decides to shave off 10°C for whatever reason and set its target to 100°C, as visible in the reply to M105: 

    ok T:26.3 /0.0 B:37.1 /100.0 @:0 B@:127

Note how it says `/100.0` instead of `/110.0`.

How the firmware behaves here internally is nothing OctoPrint can do anything about (and I doubt that rolling back to 1.2.13 will show a chance in behaviour, it might still be interesting to verify that), see also https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-is-telling-my-printer-to-do-wrong-things. I'd check the firmware configuration and EEPROM settings.

WOAAP, 2016-08-12:
So the rollback to 1.2.13 did not change anything. It still caps the temperature at 100°C. So we are on checking the firmware - despite not quite sure how to do this, yet.

nophead, 2016-08-12:
I would try sending it as an integer in case the firmware is confused by
the ridiculous floating point precision.

foosel, 2016-08-15:
Not a bug in OctoPrint, closing. Feel free to continue discussion though.

closed 2016-08-15 by foosel