OctoPrint/OctoPrint #1447: Web server always logging out users

Eriobis, 2016-08-13:
#### What were you doing?

Watching my print as usual

#### What did you expect to happen?

See the graph and the status of the print

#### What happened instead?

The browser stops refreshing because I'm logged out

#### Branch & Commit or Version of OctoPrint

Version: 1.2.15 (master branch)

#### Printer model & used firmware incl. version

#### Browser and Version of Browser, Operating System running Browser

Google Chrome	52.0.2743.116 (Official Build) m (32-bit)

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s) showing the problem:

I have read the FAQ.

foosel, 2016-08-13:
Incomplete, please provide:

> **Link to octoprint.log**
>
> **Link to contents of terminal tab or serial.log**
> 
> **Link to contents of Javascript console in the browser**

foosel, 2016-08-28:
Requested information still not provided. Closing. Request a reopen once you have provided everything asked for.

closed 2016-08-28 by foosel