OctoPrint/OctoPrint #1449: Load and print button does not run GCode in PrintStarted event

voglster, 2016-08-14:
#### What were you doing?
Printing with an event in the configuration set to run gcode on PrintStarted
clicking the button in the files list labeled "load and print"
```
events:
  enabled: True
  subscriptions:
  - event: PrintStarted
    command:
    - M80
    - G28XY
    - G0Y110F3000
    - M140S80
    - M104S150
    - M190S80
    - G28
    - G29
    - G0X10Y110Z0.5
    type: gcode
```

#### What did you expect to happen?
The gcode to run before the print started.

#### What happened instead?
The gcode did not run, instead the print just started (not turning on the printer and homing)

#### Branch & Commit or Version of OctoPrint
Version: 1.3.0.dev1232+gc6a8e2b (devel branch)

#### Printer model & used firmware incl. version
Self built i3 derivative
Marlin RC Bug Fix branch commit dcd26b0403485b844496b9b1e6767e56507d22b2

#### Browser and Version of Browser, Operating System running Browser
Chrome 52.0.2743.116 (Official Build) m (64-bit)
Windows 10 Pro (fully updated)

#### Link to octoprint.log
don't think its relevant but will provide if needed

#### Link to contents of terminal tab or serial.log
don't think its relevant but will provide if needed

#### Link to contents of Javascript console in the browser
don't think its relevant but will provide if needed

#### Screenshot(s) showing the problem:
don't think its relevant but will provide if needed

I have read the FAQ.

foosel, 2016-08-15:
I sadly cannot reproduce this on any of current `maintenance`, current `devel` or your specific version of `devel` (which is around 50 commits behind).

This is what I put into `config.yaml`:

```
events:
  subscriptions:
  - command:
    - M117 Test of ticket 1449
    enabled: true
    event: PrintStarted
    type: gcode
```

And this is what scrolled through at the start of the print, as is to be expected:

[image]

Programmatically there is no difference between clicking "Load and Print" or first loading and then manually clicking the print button, the event gets fired in the actual backend code that takes care of streaming the file to the printer.

For your specific use case I'd also rather suggest to put the code you want to run into Settings > GCODE Scripts > Before print job starts, or even your slicer (at least the temperature specific parts).

It might be possible to figure out what's wrong if you provided the logs (please *always* provide them, they are ALWAYS relevant, even if only to verify that they do not contain anything off).

voglster, 2016-08-15:
Quite strange... my auto update seems to have stopped even though i have it installed/enabled etc
[image]

[image]

Ill figure that one out later and manually update via command line... then ill run some simple tests and get the logs added here later on today.

voglster, 2016-08-15:
updated to latest commit now: edf83e498400a0079d70e61d2b0ccfa8d5d3eac9
I've tested and tested on latest version and cannot reproduce the problem on my side... closing.  If it crops up again ill grab the logs.

closed 2016-08-15 by voglster

foosel, 2016-08-15:
Perfect, thanks for reporting back!