OctoPrint/OctoPrint #1450: Octopi is Offline When Connecting with Mobile Phone, Computer and Raspi Screen

ecagatay, 2016-08-15:
#### What were you doing?

I try to connect with computer, raspberry pi screen and mobile phone at the same time. Webcam is open and touchUI plugin is installed.

#### What did you expect to happen?

I want to use Octopi interface with all device at the same time.

#### What happened instead?

Especially, when I play my finger in front of the webcam. Octopi is offline after 3 seconds.

#### Branch & Commit or Version of OctoPrint

1.2.15

#### Printer model & used firmware incl. version

My own printer. I didn't connect Raspberry Pi to Smootieboard. I just use Raspberry Pi 3 and Octopi.

#### Browser and Version of Browser, Operating System running Browser

Raspberry Pi3 : Epiphany
Windows: Firefox
Mobile Phone: Chrome

I have read the FAQ.

foosel, 2016-08-15:
> Especially, when I play my finger in front of the webcam. Octopi is offline after 3 seconds.

This sounds like your network throughput is too bad to accomodate both a webcam stream and a constant client-server-connection. This is not a bug in OctoPrint or OctoPi but an issue with your local environment. Make sure you are using at least a solid 2.5A power supply with your Pi3 (it needs more power than the older models), try moving your Pi closer to your WiFi router (the Pi3's on board WiFi antenna is very small and hence reception quality can vary depending on your individual setup) or even try a dedicated WiFi dongle or a LAN cable.

The **bug tracker** is the wrong place for this kind of **support** questions though, please seek support on [the mailinglist](https://groups.google.com/group/octoprint) or the [G+ Community](https://plus.google.com/communities/102771308349328485741) as stated in the [contribution guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#issues-tickets-however-you-may-call-them).

closed 2016-08-15 by foosel

ecagatay, 2016-08-15:
Thank you for answering. I solved my problem :)