OctoPrint/OctoPrint #1452: Error updating Octoprint from 1.2.10 to 1.2.15

Nurgak, 2016-08-16:
#### What were you doing?

I clicked on the _Update now_ button.

#### What did you expect to happen?

I expected the system to update itself to 1.2.15 master

#### What happened instead?

An error occurred:

    octoprint.plugins.softwareupdate - ERROR - Update of octoprint can not be performed

I might suspect an internet connection speed issue, maybe the latency is to blame, but I am not sure.

#### Branch & Commit or Version of OctoPrint

1.2.10 master

#### Printer model & used firmware incl. version

N/A

#### Browser and Version of Browser, Operating System running Browser

N/A

#### Link to octoprint.log

https://gist.github.com/Nurgak/a0351076942b2180d3f5e51bba3b5be2

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

N/A

foosel, 2016-08-16:
Looks indeed like a connectivity issue:

```
Reading https://pypi.python.org/simple/

2016-03-18 08:30:56,162 - octoprint.plugins.softwareupdate - ERROR - Update of octoprint can not be performed
```

Gets stuck on trying to read the package index in order to find and install new dependencies, then runs into a timeout when doing that. Check your connection, especially since you also have this in your log:

```
2016-03-18 08:17:32,216 - octoprint.plugins.pluginmanager - ERROR - Could not fetch plugins from repository at http://plugins.octoprint.org/plugins.json: ('Connection aborted.', gaierror(-2, 'Name or service not known'))
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.10-py2.7.egg/octoprint/plugins/pluginmanager/__init__.py", line 529, in _fetch_repository_from_url
    r = requests.get(repository_url)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/api.py", line 69, in get
    return request('get', url, params=params, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/api.py", line 50, in request
    response = session.request(method=method, url=url, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/sessions.py", line 465, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/sessions.py", line 573, in send
    r = adapter.send(request, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/requests-2.7.0-py2.7.egg/requests/adapters.py", line 415, in send
    raise ConnectionError(err, request=request)
ConnectionError: ('Connection aborted.', gaierror(-2, 'Name or service not known'))
```

which sounds like your Pi can't even resolve the address of the plugin repository at startup time. You need to solve your networking issues first before you can expect reliable updating or other interactions with the outside world from your setup. If you are connected via WiFi, try if moving your machine closer to the router helps, or try a cabled connection. Also make sure you have a reliable power supply for your Pi (2.5A is a good rule of thumb). If you are using a USB WiFi dongle, you could also play with an USB extension cord, maybe that helps getting better reception. Check that your machine can reach the outside, and with good reliable speeds, before proceeding.

This is not a bug though, and if it was it would not be caused by OctoPrint (which is a user space program that definitely has no influence on your network reliability but depends on running in a working environment), rather a support request, so this is the wrong place for this. Better suited are the [mailinglist](https://groups.google.com/group/octoprint) or the [G+ community](https://plus.google.com/communities/102771308349328485741), so I'm closing this and would suggest taking this there.

PS: The bot tagged your ticket because you didn't copy the *full* template.

closed 2016-08-16 by foosel

Nurgak, 2016-08-16:
I found the issue while [forcing an update](https://github.com/foosel/OctoPrint/wiki/FAQ#updating-octoprint), I got an error saying `CERTIFICATE_VERIFY_FAILED`.

In the error log above the date seemed wrong indeed (`2016-03-18 08:30:56`), I didn't think it was an issue, but the certificates were probably not valid at these times.

Setting the proper date with `sudo date -s "16 AUG 2016 08:22:00"` fixed the issue and now I'm running 1.2.15.