OctoPrint/OctoPrint #1453: webcamDaemon script on raspberrypi wiki page is incorrect

spiffomatic64, 2016-08-16:
#### What were you doing?

Following the wiki document

#### What did you expect to happen?
Run the script "/home/pi/scripts/webcamDaemon" without errors

#### What happened instead?
the script stars but shows an error when finishing: 
/home/pi/scripts/webcamDaemon: line 22: /home/pi/mjpg-streamer/mjpg-streamer-experimental: Is a directory

#### Branch & Commit or Version of OctoPrint

Version: 1.2.15 (master branch)

I was able to resolve the issue by copying lines 20 and 21 from https://gist.github.com/rjlzaitz/78f42d4baabfccb65ac6

foosel, 2016-08-16:
Not a bug but a documentation error on a wiki page editable by everyone with a Github account - it would have been faster to just fix this yourself than actually opening an issue for that and making me fix it :)

closed 2016-08-16 by foosel

spiffomatic64, 2016-08-16:
lol sorry, now I know :) 
"The more you know" lol