OctoPrint/OctoPrint #1455: [Request]Option to set the webcam's displayed image size or collapse it like the custom control button section

ntoff, 2016-08-18:
I like to set my webcam to quite a large image for super awesome timelapses but then it takes up a lot of room on the controls page so it'd be handy if you were able to set the size in the control page to be somewhat smaller. I do realise this doesn't actually change the size of the image coming out of mjpeg_streamer, nor does it do anything to cut bandwidth usage. 

Adding 'width="320"' to the webcam image section of control.jinja2 does work but then it's stuck at a maximum width of 320. So I was wondering is it possible / difficult to make the webcam image's section similar to the custom control section where it can be expanded and contracted with the difference being it doesn't completely disappear when contracted but just displays a smaller image? Again I realise this doesn't change the size of the incoming image  or cut down on bandwidth, that isn't my goal, the goal is just to not have half my screen taken up by the webcam image requiring me to scroll down a bunch to get to all the (custom) buttons.

Salandora, 2016-08-18:
Well I do fully support your request xD.
I also don't like the size of the webcam.
If @foosel is okay with it I would add a width and height setting to the settings page.
And additionally would use jquery ui's resizable feature to make it dynamically adjustable. E.g. if you by anychance want to make it bigger or smaller just for a little time but don't save these changed size.

ntoff, 2016-08-18:
well I'm glad you're on it because since making the request I've been *trying* to figure out how to do it myself (so I'm not a total leech) but I just can't wrap my head around it all :(

I was thinking that since it supports mouse over for the keyboard controls then a simple mouse hover would expand it to the full size but that probably wouldn't work on a mobile device so didn't really give that too much thought.

Salandora, 2016-08-18:
Well the resizeable plugin of jquery ui is not mobile friendly too. Sadly...
But it's at least a starting point.

foosel, 2016-08-18:
Whatever you do, make sure it a) doesn't break the keyboard control overlay and b) have in mind that it won't make it into 1.3.0 (but possibly into 1.3.1+) because unless we feature freeze that now it will take even longer to get finalized.

Salandora, 2016-08-18:
Well you know I don't mind if you release this feature with 1.3.0 or 1.20.4 as long as it gives you time to concentrate on the important things of the development. ;-)

foosel, 2016-08-18:
Just wanted to make sure ;)

ManuGithubSteam, 2016-11-25:
I would also like this feature but to make the webcam feed bigger :-)

ntoff, 2018-04-24:
Closing since it's no longer an issue and for small screens the touchUI plugin is the best way to go.

@ManuGithubSteam if you haven't already found it https://plugins.octoprint.org/plugins/fullscreen_webcam/ is able to make the web cam full screen.

closed 2018-04-24 by ntoff