OctoPrint/OctoPrint #1456: [REQUEST] Ability to send GCode from anywhere in OctoPrint

W3DRK, 2016-08-19:
It would be nice if there were a little input field uniformly positioned at the top of every screen in OctoPrint that provides the ability to send GCodes to the printer without having to navigate to the terminal screen.

For example, if I'm monitoring the video feed in the Control screen and I need to increase the brightness of my effector LEDs, I could just send M42 S196 without having to navigate to the terminal screen to send the command and then back to the control screen.

This is a feature in Duet Web control that I use all the time and I feel like it would be super useful in OctoPrint as well.  For those not familiar with the Duet's web interface, I've attached a picture to illustrate what I mean--note the Send GCode field at the top of the display.

[image]

calonmerc, 2016-08-26:
I think this functionality would be nice as well, I'm not sure if it could be accomplished with a plugin or not, but I know there are some users that would never use this.

ntoff, 2016-08-26:
I wonder if you can make a plugin put a full on serial output + gcode entry box over on the left where the status and files are listed. That'd be awesome for me as I much prefer [pronterface's layout](https://cloud.githubusercontent.com/assets/9030085/18015760/62ce3af2-6c0c-11e6-8e5f-88c82f8c5868.png)
 where everything is accessible all at once. I'm sick of flipping between tabs on my  tablet (not so bad on the PC). I've been looking at all the octoprint docs on how to write plugins but it's all way above my head :(

W3DRK, 2016-08-26:
I'm happy to see some interest in this, I was beginning to think I may be the only one who likes to have the ability to easily send commands directly to the printer rather than click around the GUI...

foosel, 2016-08-29:
As a quick overview of what you'd need from a plugin to implement this (it's not much):

  * Server side:
    * [TemplatePlugin](http://docs.octoprint.org/en/master/plugins/mixins.html#templateplugin) mixin providing a ``sidebar`` template containing a text input and possibly a send button.
    * [AssetPlugin](http://docs.octoprint.org/en/master/plugins/mixins.html#assetplugin) mixin to provide a custom view model bound to the aforementioned sidebar component.
    * Bonus: have the template config state to only make the sidebar panel visible if the user is logged in (otherwise they won't be able to send commands since the server will deny it anyway)
  * Client side:
    * Custom view model, requesting "controlViewModel" dependency
    * contents of input field bound to view model variable
    * on button click (or enter), send the entered GCODE command to the server via controlViewModel.sendCustomCommand

mtowara, 2016-08-31:
This sounds like a neat idea and I might actually be able to implement that.
*grab*

foosel, 2016-08-31:
@mtowara marked as grabbed :)

mtowara, 2016-09-01:
I put something together, it would be nice if some of you could test while I prepare the plugin repository release.

- I hooked into the terminal tab so the history scroll feature (key up/down) also works in the sidebar. 
- Commands will be automatically uppercased by the terminal send routine. Multiple commands can be seperated with semicolon.
- Controls are only enabled when logged in and connected to printer.

Get it here:
https://github.com/mtowara/OctoPrint-Gcodebar
You can install directly from OctoPi by pasting https://github.com/mtowara/OctoPrint-Gcodebar/archive/master.zip into the from URL field in the Plugin Manager dialog.

foosel, 2016-09-02:
Should now be available in the repository, I'll consider this resolved then =) Big thanks and kudos to @mtowara!

closed 2016-09-02 by foosel