OctoPrint/OctoPrint #1458: [Request] Option to leave heat on after print finishes

chaseacton, 2016-08-20:
I'd love the ability to leave the hotend/bed on for 10 minutes or so after a print so that I have time to get to the printer and start the next print without waiting for it to heat up again. I understand it could be a safety issue, but I think it would be simple to implement. After a print finishes, start a timer to turn off the hotend/bed in 10 minutes. Any user input before then would cancel the timer and leave everything heated.

ntoff, 2016-08-23:
This is more your end gcode's area, remove the M140 S0 command from your end gcode script for the bed and M104 S0 for the hot end. Be aware that leaving your hot end at printing temperature and not printing can char the filament inside if left for too long and cause issues so I don't recommend that.

chaseacton, 2016-08-24:
Good to know, thanks. It seems like this is a two-part solution then. Me removing M140 from my gcode and OctoPrint cooling the hotend 10 minutes after a finished print. Is this functionality possible via a plugin?

markwal, 2016-08-24:
Yes, there is an event for print finished and a plugin receiving that event could start a timer and then send the M140 S0 command afterward.  Trickier would be to turn off the timer on any user action, but one simple thing would be to listen for the print started event and clear the timers then.

kkoksvik, 2016-08-26:
I would love this feature, but I would not trust som one off(trigger) check or plugin handling this for safety reasons. Rather firmware or OctoPrint continually monitoring how long ago the printer was actively printing.

foosel, 2016-08-26:
@kkoksvik plugins are not more or less safe than core OctoPrint functionality. A timer started by a plugin and monitoring printer activity would be *exactly* the same as a timer started by OctoPrint and monitoring printer activity.

kkoksvik, 2016-08-26:
@foosel Shure, i can see your point, but when it comes to safety I feel it is more appropriate to be in core, as future updates could (i guess, certainly in other projects(eg. Firefox)) brake this functionality between core and plugin. Best case for safety would in my opinion be to have this in firmware, but that could lead to a lot of incomparability.

For safety concerns I do think that timing the last action of the printer and shutting the heatbed of with a one of signal is inadequate, I would rather see a continuous monitoring process handling this and other safety consernes.

Thanks a lot for the fine work you do, :-)

calonmerc, 2016-08-26:
i don't see any reason this would have any issues being controlled by a plugin, it's just going to send an M140 S0 command to the printer over the connection.... so unless you're not using a printer with standard gcode or the gcode changes... which would break ALL of octoprint, regardless if it was built in or a plugin..... it should be something optional. For instance, I would not want this, so why bother installing it? Core should be things used by every user, not just some users.

ntoff, 2016-08-26:
Here's an idea sort of related but sort of not, @foosel how hard would it be to put in a "disable heaters / motors after X minutes of inactivity" option? Let's say for instance you manually turned on the bed to pre-warm it and then something happens, you fall asleep, forget about it, etc if there's no print going on and no incoming terminal input how hard would it be to then shut the printer down?

kkoksvik, 2016-08-26:
@ntof How do you manually turn on the bed? Do you do it from OctopPrint? The only way I am able to do this is by starting a print and pausing it.

ntoff, 2016-08-27:
in the temperature graph screen there's two drop downs, one for the bed and one for the hot end.

foosel, 2016-08-29:
@ntoff it could be done easily as a plugin. Same as the original requirement. Utilizing APIs that cannot change anyhow since they would break core components (and core plugins, more and more parts of OctoPrint are slowly being migrated to being implemented as plugins, makes maintenance way easier). 

Basic approach: On server start also start a [RepeatedTimer](http://docs.octoprint.org/en/master/modules/util.html#octoprint.util.RepeatedTimer), triggering for example every minute. Have it do basically this:

  * is the printer [currently not operational](http://docs.octoprint.org/en/master/modules/printer.html#octoprint.printer.PrinterInterface.is_operational) => nothing to do
  * are all target [temperatures](http://docs.octoprint.org/en/master/modules/printer.html#octoprint.printer.PrinterInterface.get_current_temperatures) off => nothing to do
  * is the printer currently [printing](http://docs.octoprint.org/en/master/modules/printer.html#octoprint.printer.PrinterInterface.is_printing) or [paused](http://docs.octoprint.org/en/master/modules/printer.html#octoprint.printer.PrinterInterface.is_paused) => nothing to do
  * do we have a countdown: decrease it
    * has it reached 0? => [disable all heaters](http://docs.octoprint.org/en/master/modules/printer.html#octoprint.printer.PrinterInterface.set_temperature)
  * if we come here, no countdown situation yet, start countdown, e.g. by setting internal counter to 10

More fancy versions would also hook into the sent commands (via the corresponding [hook](http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-comm-protocol-gcode-phase)) and only start the countdown if nothing but the regular `M105` polls (whitelisting approach) or no movement or temperature commands (blacklisting approach) were sent, immediately disabling the countdown again if that changes.

But I also agree that something considered a safety feature should actually be built directly into the firmware (especially considering how easily the communication to the printer can go to a broken state thanks to there not being a properly defined protocol). I know that at least some firmware versions actually do that too

kkoksvik, 2016-08-30:
@ntoff I do not get any feedback when setting the temperatures as you describe, the graphs do not update. I am running repetier 0.91 (I should probably try to upgrade). I also have had some strange things happening with the temperatures changing without me knowing why, even turning off. Is all features working well for you when changing temperatures this way?

@fossel, @ntoff 
I do not think that checking operational and temperature status is necessary, checking for these statuses could introduce more opportunity for failure if for some reason the statuses are corrupted in any way. Sending the heater off commands every 10 minutes should not be any problem even if not operational or temperatures is off already.

During paused prints I have experienced that temperatures have been turned off, is there intensionally put in some kind of max pause time? 
If not, could it be a good idea to have separate timeouts for paused prints?

foosel, 2016-08-30:
> Sending the heater off commands every 10 minutes should not be any problem even if not operational or temperatures is off already.

Printer not operational == no connection to printer, hence no way at all to send something to it. But yes, temperature check could be left out too. But I can only already hear a bunch of people shouting "why is it sending these commands when it's already off" ;)

> I am running repetier 0.91

All options tagged "Repetier" in Settings > Features are checked? (yes, I know, I need to move that stuff to the serial tab - I need to do a lot of stuff but the day has only so many hours)

> During paused prints I have experienced that temperatures have been turned off, is there intensionally put in some kind of max pause time? 

Nope. Do you maybe have that configured under Settings > GCODE Scripts > After print job is paused? That should ship as empty.

kkoksvik, 2016-08-30:
> Printer not operational == no connection to printer, hence no way at all to send something to it. But yes, temperature check could be left out too. But I can only already hear a bunch of people shouting "why is it sending these commands when it's already off" ;)
Guess, better safe than sorry. You could send them to this link [https://en.wikipedia.org/wiki/Fail-safe]() 

> All options tagged "Repetier" in Settings > Features are checked?
Wow, that does it :-) 

> Nope. Do you maybe have that configured under Settings > GCODE Scripts > After print job is paused? That should ship as empty.
It is empty, and it feels more intermittent. I will monitor it and see if it happens after i switched the Repetier features.

foosel, 2016-08-30:
> You could send them to this link https://en.wikipedia.org/wiki/Fail-safe

Before I started out on this project, I still believed people would read provided informational material and listen to reason. I've since rectified this assumption ;)

ntoff, 2017-12-04:
@foosel Does this plugin http://plugins.octoprint.org/plugins/heatertimeout/ maybe fulfil the needs of this request? Perhaps it can now be closed? If a user wants the heaters left on, he or she can install that plugin and then remove any heater commands from their ending gcode, and let the timer plugin take care of turning them off.

foosel, 2017-12-04:
@ntoff good point, thanks!

closed 2017-12-04 by foosel