OctoPrint/OctoPrint #1459: Before Print Start scrip not running

KC703, 2016-08-23:
Please read the "guidelines for contributing" that are linked ^-- just
up there. Also read the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions). Mark requests with
a [Request] prefix in the title please. Fully fill out the bug reporting
template for bug reports.

Do not seek support here ("I need help with ..."), that belongs on
the mailing list or the G+ community (both linked in the "guidelines
for contributing" linked above, read it!), NOT here.

Thank you!

----

#### What were you doing?

Using "Before Print" Script to preheat bed before starting print.  Script consists of: M190 S85

Its worked in the past, up to about a month ago.  Reimaged my SD with most recent, then switched to the devel track: 
Version: 1.3.0.dev1325+g4ff869c (devel branch)

#### What did you expect to happen?
The printer should preheat the bed to 85C, then start the selected gcode file.  Upon starting the print, hotend and bed should be set to temperatures specified from slicer (240C, 110C respectively)

#### What happened instead?
Printer ignored preheating of bed, and set extrusion and bed temps specified in project gcode.

#### Branch & Commit or Version of OctoPrint
DEVEL, Version: 1.3.0.dev1325+g4ff869c (devel branch)

[Can be found in the lower left corner of the web interface.]

#### Printer model & used firmware incl. version
Lulzbot TAZ5, Marlin

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser
Chrome Version 54.0.2832.2 dev-m, Windows 10
[If applicable, always include if unsure.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. Always include and never truncate.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.]

#### Screenshot(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

foosel, 2016-08-24:
Please provide a *completely filled out* ticket template, including all requested logs. That stuff is requested for a reason ;)

KC703, 2016-08-25:
Updated with more info... still need upload of the temp graph on start of print.

KC703, 2016-08-25:
For fun, I switched back to the "Master".  And the Before Print script worked.  Here's a picture of the Temp Chart.

[image]

KC703, 2016-08-25:
Here's the Temp Graph when starting a print in the "Devel" track.  The "Before" script specified to preheat the bed is bypassed.

[image]

I'll switch back to the Master for the time being.

foosel, 2016-08-25:
Ok, found it. That was a bug introduced in 5bf5e336eca92b6b52aa1198ea047bea74fefcd1 by an incomplete move of functionality. All other scripts were properly called in the code, just not the `beforePrintStarted` one.

Fixed in 4fe944eb85181118159ed8d1339c2f6db9433ce5 (at least I cannot reproduce it anymore). Can you update back to current `devel` and verify?

foosel, 2016-08-25:
Btw, the `serial.log` is empty unless you enable it. The contents of the terminal tab however - why not necessarily a full log of what happened due to content already having scrolled out - can still help, which is why the ticket template asks for that first place :)

KC703, 2016-08-25:
Looks good!  Before script working on the DEVEL track.

Thank you!
Ken

closed 2016-08-25 by KC703

KC703, 2016-08-25:
Definitely understand the need for logs to understand and troubleshoot.  It's not readily apparent where to enable the serial.log... maybe a note in the Github issue template to specify where to turn it on in the "Serial Connection" tab.    :-)

Thanks again for the quick resolution.