OctoPrint/OctoPrint #1460: Error: Failed to autodetect serial port, please set it manually.

pauloesperon, 2016-08-23:
> Error: Failed to autodetect serial port, please set it manually. <

I'm getting this error while trying to connect with my Flashforge Dreamer.
I flashed Octoprint successfully and I can connect to the pi via WiFi. The camera is also working and displaying image on Octoprint. 

There seems to be a connection problem between the Pi and the printer. I tried with different cables and tried connecting them in different ports on the pi with no success. 

Version: 1.2.15 (master branch)

Printer: Flashforge Dreamer
Firmware: V2.2 20150107
Chrome Version 52.0.2743.116 m

Any ideas what might be causing the problem?

pauloesperon, 2016-08-23:
Error: Failed to autodetect serial port, please set it manually.

Any ideas what might be causing the problem?
#### What were you doing?

I'm getting this error while trying to connect with my Flashforge Dreamer.
I flashed Octoprint successfully and I can connect to the pi via WiFi. The camera is also working and displaying image on Octoprint. 

There seems to be a connection problem between the Pi and the printer. I tried with different cables and tried connecting them in different ports on the pi with no success. 

#### What did you expect to happen?
To connect with the printer
#### What happened instead?
I got this error: Error: Failed to autodetect serial port, please set it manually.

#### Branch & Commit or Version of OctoPrint
Version: 1.2.15 (master branch)

#### Printer model & used firmware incl. version
Printer: Flashforge Dreamer
Firmware: V2.2 20150107

#### Browser and Version of Browser, Operating System running Browser
Chrome Version 52.0.2743.116 m

#### Link to octoprint.log

not sure what this is

#### Link to contents of terminal tab or serial.log

not sure what this is

#### Link to contents of Javascript console in the browser

[not sure what this is

#### Screenshot(s) showing the problem:
[image]

markwal, 2016-08-24:
Nobody has made a plugin for the FlashForge Dreamer yet as far as I know.   See #1263 .

closed 2016-08-24 by markwal