OctoPrint/OctoPrint #1461: Change Printer profile confirm button hangs endlessly

averter, 2016-08-27:
#### What were you doing?

I wanted to change the speed of extrusion and save this into the default printer profile.

#### What did you expect to happen?

After hitting the confirm it should after a few seconds have returned to the normal Octoprint web interface.

#### What happened instead?
The confirm change button hangs and keeps showing the saving circle endlessly. I've waited at least 10 minutes before giving up. After aborting no matter which profile I try to load the button always shows the same stats and the extruder speed is not updated

#### Branch & Commit or Version of OctoPrint
     Version: 1.3.0.dev623+g4fcb853 (devel branch) 

#### Printer model & used firmware incl. version
Prusa i3 reprap Marlin R6

#### Browser and Version of Browser, Operating System running Browser
Firefox 48

#### Link to octoprint.log
https://gist.github.com/averter/2e478cb571c6266c627ed6289526660a

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser
[image]

#### Screenshot(s) showing the problem:
[image]

I have read the FAQ.

foosel, 2016-08-27:
Your devel version is ancient (there have been more than 600 commits since then). Do you still see the same behaviour on current devel?

foosel, 2016-08-27:
Also something went wrong with your screenshot of the js console, please provide that if you can still reproduce the issue on current devel.

averter, 2016-08-27:
I thought that Octoprint automatically updated the current version. After updating manually to the current devel works perfectly. 
I still had a similar issue when changing the default printer though. The star didn't moved from one printer to the other unless I deleted the profile I didn't wanted.

Thanks

closed 2016-08-27 by averter

foosel, 2016-08-27:
It updates release versions on its own, for development versions you have to configure it differently. Not sure if that was already part of the ui of your older version. Switch to commit tracking on the current one in the software update settings (nutrition in the upper right corner)