OctoPrint/OctoPrint #1463: UI buttons donDirect upload to .octoprint\uploads

JosvdP, 2016-08-28:
TLDR; Buttons for Remove, Load and Load and Print don't work when uploading new GCODE directly in Octoprint Uploads folder. Normally uploading through UI works as normal.

#### What were you doing?
- Upload new .gcode directly from Simplify3D into a network folder that is \\printer\pihome\.octoprint\uploads
- New file appears under "Files" in OctoPrint
- I try clicking the buttons under the file that just appeared: Remove, Load, Load and Print.

#### What did you expect to happen?
I expect the buttons to either remove, load or load and print the file.

#### What happened instead?
Nothing happens! However, I checked the Chrome developer tools Console and there I see a javascript error (see below)

#### Branch & Commit or Version of OctoPrint
Version: 1.2.15 (master branch)

#### Printer model & used firmware incl. version
BigBox Dual
Marlin 1.1.0-RC3

#### Browser and Version of Browser, Operating System running Browser
Windows 10
Chrome 52.0.2743.116 m (64-bit)

I also replicated the bug in FireFox 48.0.2

#### Link to octoprint.log
http://pastebin.com/4quz0Grw

#### Link to contents of terminal tab or serial.log
Nothing there?

#### Link to contents of Javascript console in the browser
```
packed_libs.js?35d69079:5 DELETE http://printer/api/files/local/2cm%20cube.gcode 404 (NOT FOUND)

send @ packed_libs.js?35d69079:5
ajax @ packed_libs.js?35d69079:5
GcodeFilesViewModel.self.removeFile @ packed_app.js?77633f61:1134
(anonymous function) @ VM2461:3
(anonymous function) @ packed_libs.js?35d69079:1020
dispatch @ packed_libs.js?35d69079:4
y.handle @ packed_libs.js?35d69079:4
```

#### Screenshot(s) showing the problem:
[image]

I have read the FAQ.

foosel, 2016-08-28:
That's a duplicate of #1434. See that for why that is not really a bug and the measures taken to resolve it anyhow in the next release.

closed 2016-08-28 by foosel

JosvdP, 2016-08-28:
Ah, I see! Didn't know it was related to special characters so it didn't show up for me when I searched before reporting. Sorry!

foosel, 2016-08-28:
No harm done :)