OctoPrint/OctoPrint #1466: Octoprint does not interpret the Gcode correctly!

connected201, 2016-08-29:
Hi, everyone, i generated Gcode in simplify3d, and upload in octoprint, so, as you can see, on the photo, i made 2 print, and 2 fails, 
as you can see in the photo, octoprint normal, approximately 6mm, after print below 45 degree angle!
printer: prusa mendel i3.

i atach Gcode and photos of bad print, best regards
[X-End.zip](https://github.com/foosel/OctoPrint/files/442154/X-End.zip)
[image]
[image]
[image]
[image]
[image]
[image]

foosel, 2016-08-29:
That's a classical layer shift issue, caused by losing steps on one of your axes. That is a (probably mechanical) problem with your printer, not OctoPrint.

OctoPrint instructs your printer's firmware to execute commands from the GCODE files you print through OctoPrint. It expects your firmware to execute those commands correctly. In turn, your firmware expects your printer's mechanics to move correctly. You have a problem between the firmware and your printer mechanics here. You need to solve this before you will be able to print from ANY host.

closed 2016-08-29 by foosel

connected201, 2016-08-29:
i dont think, because, i made 2 print probes!, so, this problem made in the same place! mechanical problem excluded! right now, i take a print from simplify3d

foosel, 2016-08-29:
Again. All regular OctoPrint does is send stream the lines from your GCODE file to your printer. It doesn't post process them in any way, it doesn't change them before sending them to the printer. Something like the above (a gradual shift along one axis of consecutive layers, starting at a specific height) would be pretty much impossible to even do on purpose in OctoPrint. Your printer is losing steps there, on one axis - X or Y. That might be due to mechanical issues (slipping belt, loose pulley, ...), or due to overheating or similar problems. [Read this on that topic](https://www.simplify3d.com/support/print-quality-troubleshooting/#layer-shifting-or-misalignment). Also [this FAQ entry](https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-is-telling-my-printer-to-do-wrong-things).

IF you have a plugin installed that does in fact modify the GCODE from the file before it's been sent out, it could be a malfunction or wrong configuration of that plugin causing the generated output to be shifted (e.g. I'm not sure if the GPX plugin by @markwal could produce such artefacts when a wrong printer profile is selected, maybe he can shed some light on that). But - again - stock OctoPrint can't and doesn't do that and also doesn't ship with any such plugins.

But feel free to seek support for this issue on the mailing list or on the G+ community (both linked in the contribution guidelines that the bot linked for you up there and which you should definitely read since they also tell you what a bug report should contain to be processable at all - your's lacks almost everything needed to even look into it).

hungerpirat, 2016-08-29:
@connected201 clearly a mechanical problem. check your printer

connected201, 2016-08-29:
OK, i will check if after this print i have same problem!

connected201, 2016-08-29:
so, problem is in generated Gcode!
But!, any program to view Gcode, show that Gcode is OK, all objects are ok, where is problem?

foosel, 2016-08-29:
[**Check. Your. Printer.**](https://www.google.de/#q=3d+print+layer+shift)

It's not your GCODE. From your pictures, it's a classical layer shift caused by some hardware or electronics issue, NOT anything in software. Which we've been trying to tell you for hours now. You are barking up the wrong tree here. Check your printer's belts, pulleys, stepper settings, free axes movement, heat dissipation, and so on.

connected201, 2016-08-29:
i believe you, but, i print right now another pieces, and all ok!, printer are new, belts are ok, i know what is layer shift! but, i think its' not layer shift...i give 10$ for who print from my Gcode, and give me photo for success results! :)

markwal, 2016-08-29:
@connected201 one way that it can be the gcode is if you tell the slicer to over-extrude. Then the infill in some spot that is getting fully filled will pile up and if you're "lucky" the nozzle will hit it and if your bed adhesion is good enough, you'll skip steps. But like everybody says, nothing to do with octoprint.

Salandora, 2016-08-29:
@connected201 what speed settings do you use?
What acceleration settings?
Maybe reduce both by half and try again.

Salandora, 2016-08-29:
Sorry forgot, what electronic do you use? What stepper drivers? At what ampere do you drive the steppers? How hot are the drivers while printing?
It also could be a faulty driver or a drover which needs to deliver to much amps and overheats.