OctoPrint/OctoPrint #1468: [Bug?] M117 checksum incorrect. Loop needs iteration limit/abort.

gruvin, 2016-08-30:
#### What were you doing?

**NOTE:** This _could_ be a fault in my Marlin firmware. Seems highly unlikely though.

Attempting to print with GCODE containing ...

```
M117 Printing...
```

#### What did you expect to happen?

Print to start.

#### What happened instead?

Looking in the Terminal window, I see that Marlin is complaining that the checksum for the M117 line is errant and is requesting a re-send. OctoPrint is complying, by resending the same line again, which just gets the same error/resend response.

This loop continues forever. May I suggest some code to count how many times the same line is resent under these conditions and a full abort with error report if that count is exceeded?

I am assuming of course that Marlin is calculating the checksum for the line correctly. This assumption could be wrong?

#### Branch & Commit or Version of OctoPrint

Forked and recently merged from **master-afdb98a92fb5dae2661aa252975933be55bef2c2**

I think this is v1.2.15, though the last commit log says, "Postponing the release of 1.2.15 until tomorrow.", back in July.

Only changes I have made are to templates.

#### Printer model & used firmware incl. version

Home Made, Marlin 1.0.x based.

#### Browser and Version of Browser, Operating System running Browser

Chrome, current version (Mac).

#### Link to octoprint.log

N/A

#### Link to contents of terminal tab or serial.log

Printer in use right now, so cannot get this sorry. 

I believe the above description covers it and the bug should be very easy to duplicate, assuming it's not my Marlin firmware causing the problem.

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s) showing the problem:

N/A

Issue tempalte wrote: "I have read the FAQ."
Amen! :-P

Thanks.

foosel, 2016-08-31:
I'm gonna need a snippet of communication output with this because it doesn't reproduce here.

foosel, 2016-08-31:
Also ALWAYS provide the octoprint.log. It also contains full version numbers, plugin lists, resend request logs and is hence absolutely essential for debugging.

gruvin, 2016-08-31:
Thanks for the rapid response. Only, this one is egg on my face. The problem is in fact introduced by my own code -- in the firmware, to do with accepting lower case commands. ("Printing..." becomes "PRINTING..." just before the checksum is calculated.) It's code I wrote many months ago. D'oh!

So sorry for the trouble!

Bryan
New Zealand

closed 2016-08-31 by gruvin

foosel, 2016-08-31:
@gruvin :D Always glad when it's something simple as that :)