OctoPrint/OctoPrint #1469: Feature Request: Add "Layer (x) of (y)" to "Machine State" pane

sfsdfd, 2016-08-31:
I've noticed that 95% of the time I visit the "GCode Viewer" tab, it's for the sole purpose of seeing which layer is currently being printed and the total layer count.

Every time I hit that tab, I have to wait through both the model analysis routine, and the drawing routine to show what's being done on the current layer. While I really appreciate that functionality, it's a bit of a drag to experience a significant delay in retrieving the simple information that I seek, which is caused by functionality that's not of interest.

A very simple solution is inserting a line somewhere in the basic "Machine State" pane that reports the current and total layer counts. This could supplement or even be combined with other information that's currently in the pane, like the "Printed: (x) MB / (y) MB" field.

markwal, 2016-08-31:
Duplicate of #1144 among [many others](https://github.com/foosel/OctoPrint/search?q=total+layer&type=Issues&utf8=%E2%9C%93)

markwal, 2016-08-31:
And that issue points at this plugin: http://plugins.octoprint.org/plugins/displayz/

closed 2016-08-31 by markwal

sfsdfd, 2016-08-31:
Ah... I've committed the cardinal sin of forum posting: failing to search. Apologies, and thanks for the cleanup on aisle 4.