OctoPrint/OctoPrint #1471: Flask-0.10.1 error

orangeudav, 2016-08-31:
Traceback:
```
2010-01-01 03:01:29,989 - octoprint - ERROR - Exception on /intermediary.gif [GET]
Traceback (most recent call last):
  File "/home/cubie/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/cubie/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1479, in full_dispatch_request
    response = self.process_response(response)
  File "/home/cubie/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1693, in process_response
    self.save_session(ctx.session, response)
  File "/home/cubie/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 837, in save_session
    return self.session_interface.save_session(self, session, response)
  File "/home/cubie/OctoPrint/venv/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/sessions.py", line 326, in save_session
    val = self.get_signing_serializer(app).dumps(dict(session))
  File "/home/cubie/OctoPrint/venv/local/lib/python2.7/site-packages/itsdangerous-0.24-py2.7.egg/itsdangerous.py", line 566, in dumps
    rv = self.make_signer(salt).sign(payload)
  File "/home/cubie/OctoPrint/venv/local/lib/python2.7/site-packages/itsdangerous-0.24-py2.7.egg/itsdangerous.py", line 409, in sign
    timestamp = base64_encode(int_to_bytes(self.get_timestamp()))
  File "/home/cubie/OctoPrint/venv/local/lib/python2.7/site-packages/itsdangerous-0.24-py2.7.egg/itsdangerous.py", line 220, in int_to_bytes
    assert num >= 0
AssertionError
```

OctoPrint 1.2.15 (master branch)

https://github.com/foosel/OctoPrint/issues/149 not help for me:
```
Traceback (most recent call last):
  File "/home/cubie/OctoPrint/venv/bin/octoprint", line 5, in <module>
    from pkg_resources import load_entry_point
  File "/home/cubie/OctoPrint/venv/local/lib/python2.7/site-packages/distribute-0.6.24-py2.7.egg/pkg_resources.py", line 2711, in <module>
    parse_requirements(__requires__), Environment()
  File "/home/cubie/OctoPrint/venv/local/lib/python2.7/site-packages/distribute-0.6.24-py2.7.egg/pkg_resources.py", line 584, in resolve
    raise DistributionNotFound(req)
pkg_resources.DistributionNotFound: Flask-Login>=0.2.2,<0.3
```

closed 2016-09-28 by GitIssueBot

dmiddlecamp, 2016-11-20:
I ran into this and discovered it's because of a very old system time set on the operating system.  If you update your system time, you can fix this:

```
date --set "your current time"
```

ShacharWeis, 2017-11-10:
dmiddlecamp, I love you and want to have your babies. You just saved me ungodly amount of time.

PrecisionWaveRF, 2018-02-10:
I'm having the same issue on some devices that have run out of RTC backup battery and I can't allow console access to users to manually set the system time. How can I get around this issue? e.g. get the time from the client to set the system time or just allow an old system time?

dmiddlecamp, 2018-02-10:
One option is to add a script to run on start up to set some reasonably recent system time even if it’s inaccurate

PrecisionWaveRF, 2018-02-10:
will this work in 10years from now too?

dmiddlecamp, 2018-02-10:
Does your setup require not fixing your clock for the next decade?  ;)

PrecisionWaveRF, 2018-02-10:
actually yes :) it's independent devices that are not supposed to be accessed via shell by customers. Units are not necessarily web connected, so NTP wont work. There's no way for me to know whether the time on the device is correct other than getting the client time and comparing that to the device time. But since flask won't render due to above AssertionError I can not do that. Any idea?

dmiddlecamp, 2018-02-10:
So, real time clocks powered by watch batteries lasting 10 years are not unusual, so I would strongly push back on the lack of that being a requirement.  I think from a objective standpoint, your other choices would be to try and keep the time at least a little up to date via a variety of means, OR maybe you could patch flask to not sign the session with a timestamp?