OctoPrint/OctoPrint #1472: [Feature request]  execute layer script change BEFORE taking picture

st-f, 2016-08-31:
Pretty self explanatory - it would be really useful for me as I have my camera mounted on a moving part. Timelapses are shaking as for now as each layer start at a different point.

foosel, 2017-01-27:
Timelapses will always be shaking because the firmware has a planning buffer into which it puts movement commands which then get executed asynchronously. So in a nutshell, there's no way for OctoPrint to know when a move command it sent to the firmware might actually be executed (including a Z-layer switch) - might be immediately, might be a second later, or even longer. It all depends on what is already in the buffer and how the firmware decides to execute that.

What could be done is modifying the firmware to send an [action trigger](http://docs.octoprint.org/en/master/features/action_commands.html) on serial at a specific point in the movement - so from the planner, out of band - and writing an OctoPrint plugin that somehow turns that into a timelapse snapshot (and allowing timelapse enhancement through plugins is something I actually have on my TODO list already, to allow snapshot control from plugins).

A non-shaking timelapse could also be achieved by first waiting for the buffer to run dry just before the layer switch via `M400`, doing the snapshot and then moving on, but while that would get you nice timelapses it would probably ruin your prints or at least severly reduce their quality due to the enforced micro-pause on each layer change. I'd say that isn't worth it.

But unless someone does that firmware modification first, what you are asking for is sadly impossible.

closed 2017-01-27 by foosel

st-f, 2017-01-27:
Thanks for your detailed answer, will check this out!

WolfgangFahl, 2019-02-28:
My idea would be different. If the g-code coud be modified to move the extruder to the side on a level change than this could be optically detected with the camera. It's just a matter of calulating the difference between pictures and triggering on the extruder being in a certain position.
Was something like this already discussed somewhere?