OctoPrint/OctoPrint #1476: Will Octoprint NOT use temperatures for the hotend below 200 degrees?

Nvigr8, 2016-09-04:
Using Simplify3D, If I program the job to use a hotend temp below 200, it simply will not extrude.  Manually or otherwise.  Would someone explain this to me please?  Many thanks.

markwal, 2016-09-05:
Given the little information here, my first guess is that it is unlikely to have anything to do with OctoPrint.  Some materials need to be hotter than 200 to extrude.  What material?  What printer?  Are you able to extrude manually when using some other software (is that why you believe OctoPrint has something to do with it?)  You'll probably want to post the question along with the extra info on the google group where more people are reading and might be able to help.

closed 2016-09-05 by markwal

nophead, 2016-09-05:
Do you get a "Cold extrusion prevented" message in the terminal tab?

Marlin firmware has a minimum extrusion temperature although it would
normally be set to less than 200C. You can disable it with M302 - Allow
*cold* extrudes.