OctoPrint/OctoPrint #1478: Linear print time estimate is displayed too early

foosel, 2016-09-05:
**Problem**

Even if no other estimation sources are available, OctoPrint should only display the highly inaccurate linear estimate to the user if either 30min of the print have passed or 30% of the print job have been completed. In 1.2.15 the linear approximation is directly used instead of displaying "Calculating" until one of the above happens.

**Solution**

Make sure to actually convert passed time to minutes before comparison.

Already solved in 11bdc17, ticket created for documentation purposes.

ManuGithubSteam, 2016-09-19:
Hi
another thing that happend to me: I print a part twice, same gcode file. It could know (save) the time needed the first time of printing and then simply display that. Now it does recalculate it....

foosel, 2016-09-19:
Hm, it actually SHOULD save the print time: https://github.com/foosel/OctoPrint/blob/master/src/octoprint/printer/standard.py#L987

Did you update the GCODE file in any way or switched to another printer profile by any chance?

ManuGithubSteam, 2016-09-19:
I  just reuploaded it because i deleted to old one by accident but no
changes to the file. At least i remember none.

foosel, 2016-09-19:
ah, that explains it then. When you delete the old file, OctoPrint also throws away the metadata. It can't know if you'll reupload the file again, and if it just expected you to do that, it would use a lot of space for obsolete metadata and slowly fill up your disk with garbage. So, deletion or reupload (which is a deletion + recreation) -> metadata is reset.

foosel, 2016-09-24:
Fixed in 1.2.16 which was released yesterday. Also fixed on `devel`.

closed 2016-09-24 by foosel